platformio/platform-espressif32 #341: Cmakelists.txt suggestions for multiple components

wegunterjr, 2020-05-07:
I was able to get it included by adding this to my CMakeLists at the top level:

set(EXTRA_COMPONENT_DIRS "components/cpp_utils")

and then in the cpp_utils folder, I added this:

# Edit following two lines to set component requirements (see docs)
set(COMPONENT_REQUIRES
  "console"
  "fatfs"
  "json"
  "mdns"
  "nvs_flash"
)
set(COMPONENT_PRIV_REQUIRES )

file(GLOB COMPONENT_SRCS
  LIST_DIRECTORIES false
  "*.h"
  "*.cpp"
  "*.c"
  "*.S"
)
set(COMPONENT_ADD_INCLUDEDIRS ".")

register_component()
My question is, if I have a 2nd Library next to cpp_utils in the components folder, how does it know about cpp_utils. I tried adding it in the COMPONENT_REQUIRES , but that doesn't seem to work...

thoughts?

valeros, 2020-05-07:
Hi @wegunterjr ! Thanks for the report. First thing, you shouldn't override `EXTRA_COMPONENT_DIRS` variable as PlatformIO also relies on it so you should append new components to this variable.
I use the following configuration:

Top level `CMakeLists.txt`:
```cmake

cmake_minimum_required(VERSION 3.5)
list(APPEND EXTRA_COMPONENT_DIRS "components/hello" "components/test")
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(hello-world)
```

`CMakeLists.txt` for `hello` component:
```
idf_component_register(SRCS "hello.c" INCLUDE_DIRS ".")
```

`CMakeLists.txt` for `world` component that depends on `hello`:
```cmake
idf_component_register(
  SRCS "world.c"
  INCLUDE_DIRS "."
  PRIV_REQUIRES hello
)
```

wegunterjr, 2020-05-15:
okay..getting some different errors now relating to git issues?   Seems like a strange error?  not necessarily related, but what if I modify your suggestion to being that :
world depends on hello, but hello also depends on esp idf core libraries?

wegunterjr, 2020-05-15:
CMake Error at CMakeLists.txt:3 (list):
  list does not recognize sub-command append

valeros, 2020-05-15:
> world depends on hello, but hello also depends on esp idf core libraries?

What's the problem then? Specify required dependencies in CMakeLists.txt for `hello` component.

> list does not recognize sub-command append

That's strange, could you please attach here the `CMakeLists.txt` which causes this error? For me it looks like there is an outdated CMake somewhere in your system, but it shouldn't be a problem when you're using PlatformIO as we only install recent CMake versions.

wegunterjr, 2020-05-15:
Yeah, that's the hard part, i must be missing something.  I think I can figure out the hello world example you suggested, but what if in there I am reaching into esp idf core libraries...
in world, it has #include #include <esp_log.h>  or similar... 

This is my top level folder CMakeLists.txt
```
cmake_minimum_required(VERSION 3.16.0)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
SET(EXTRA_COMPONENT_DIRS  "components/cpp_utils" )
project(connect-esp32)

```

this is in the components/cpp_utils folder CMakeLists.txt
`idf_component_register(SRCS "BLEDescriptor.cpp" INCLUDE_DIRS ".")`
Do i just add each file i want to add to this?

Currently can't even find the main target...

```
DEBUG: Current (ftdi) On-board (ftdi) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40000.200303 (4.0.0)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.190618 (2.28.51)
 - toolchain-xtensa32 2.80200.200226 (8.2.0)
Warning! Starting with ESP-IDF v4.0, new project structure is required: 
https://docs.platformio.org/en/latest/frameworks/espidf.html#project-structure
Reading CMake configuration...
Error: Couldn't find the main target of the project!
```

this is my platformio.ini   

```
`[platformio]
default_envs = esp-wrover16

[env:esp-wrover16]
platform = espressif32
framework = espidf
board = esp-wrover16
build_flags = 
   -D PIO_FRAMEWORK_ESP_IDF_ENABLE_EXCEPTIONS -g3
monitor_speed = 115200
monitor_rts = 0
monitor_dtr = 0
board_build.partitions = partition_custom.csv`
```

```
Updating tool-unity                      @ 1.20500.200322 [Up-to-date]
Updating tool-scons                      @ 2.20501.191222 [Up-to-date]

Platform Manager
================
Platform Espressif 32
--------
Updating espressif32                     @ 1.12.1         [Up-to-date]
Updating toolchain-xtensa32              @ 2.50200.80     [Up-to-date]
Updating tool-ninja                      @ 1.7.1          [Up-to-date]
Updating toolchain-esp32ulp              @ 1.22851.190618 [Up-to-date]
Updating tool-cmake                      @ 3.16.4         [Up-to-date]
Updating framework-espidf                @ 3.40000.200303 [Up-to-date]
Updating tool-esptoolpy                  @ 1.20600.0      [Up-to-date]
Uninstalling toolchain-xtensa32 @ 2.80200.200226:       [OK]

Platform Espressif 32
--------
Updating espressif32                     @ 1.11.0         [Detached]
Updating toolchain-esp32ulp              @ 1.22851.190618 [Up-to-date]
Updating tool-esptoolpy                  @ 1.20600.0      [Up-to-date]
Updating toolchain-xtensa32              @ 2.50200.80     [Up-to-date]
Updating framework-espidf                @ 3.30300.190916 [Up-to-date]

Platform Espressif 32
--------
Updating espressif32                     @ 1.7.0          [Detached]
Updating tool-esptoolpy                  @ 1.20600.0      [Up-to-date]
Updating toolchain-xtensa32              @ 2.50200.80     [Up-to-date]
Updating framework-espidf                @ 3.30103.190221 [Up-to-date]

Platform Native
--------
Updating native                          @ 1.1.2          [1.1.3]
Uninstalling native @ 1.1.2:    [OK]
PlatformManager: Installing native @ 1.1.3
native @ 1.1.3 has been successfully installed!

Library Manager
===============
Library Storage: /root/.platformio/lib
```

valeros, 2020-05-15:
> This is my top level folder CMakeLists.txt

```cmake
SET(EXTRA_COMPONENT_DIRS  "components/cpp_utils")
```
should be 
```cmake
list(APPEND EXTRA_COMPONENT_DIRS "components/cpp_utils")
```
and must be placed above the line
```cmake
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
```

```cmake
cmake_minimum_required(VERSION 3.16.0)
list(APPEND EXTRA_COMPONENT_DIRS "components/cpp_utils")
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(connect-esp32)
```

wegunterjr, 2020-05-15:
Dang...thanks.  Got passed the can't find main, and looks like it is getting into the components folder.
I read the line in the docs, but totally spaced it.  Thanks for the assist.

`To set these variables, use the cmake set command ie set(VARIABLE "VALUE"). The set() commands should be placed after the cmake_minimum(...) line but before the include(...) line.`

wegunterjr, 2020-05-15:
So for this part, where the world component is referencing hello...
Do i need to do some special name for the cpp_utils, or does having it listed as a component make it available?
```
components/cpp_utils
components/listOfFiles
```

listOfFiles has code that references code i nthe cpp_utils folder...

components/listOfFiles  CMakeLists.txt
```
idf_component_register(
  SRCS "world.c"
  INCLUDE_DIRS "."
  PRIV_REQUIRES cpp_utils
)
```

valeros, 2020-05-15:
> Do i need to do some special name for the cpp_utils, or does having it listed as a component make it available?

> listOfFiles has code that references code in the cpp_utils folder

Yes, if it's declared as component it should be visible for other components (keep in mind that you decide what parts of `cpp_utils` are visible to other components using `INCLUDE_DIRS` option)

wegunterjr, 2020-05-19:
did this change recently with the update in platformio?   So, now just use the lib folder?  no need to specify the components folder?

valeros, 2020-05-19:
Simple libraries that don't depend on ESP-IDF build system can be placed in the `lib` folder.

wegunterjr, 2020-05-19:
@valeros  got ya.   I changed it back to components. I am seeing some progress...i think.  HA!  Might just be different libraries, but also finding that as we move to ESP 4 that some of the includes have changed.   like wpa_supplicant doesn't have a crypto/common.h.
Folder structure right now:

- boards
- components
-- cpp_utils
-- testLibrary
- src
-- main.cpp
- CMakeLists.txt
partition_custom.csv
platform.ini

wegunterjr, 2020-05-20:
so, my library that references cpp_utils seems to be 'working', but i am getting errors now for using core esp libraries... 

`fatal error: esp_spiffs.h: No such file or directory`

```
# Edit following two lines to set component requirements (see docs)
set(COMPONENT_REQUIRES
  cpp_utils
  console
  fatfs
  json
  mdns
  nvs_flash
  lwip
  esp_eth
  freertos
  mbedtls
  wpa_supplicant
  log
  spiffs
)
set(COMPONENT_PRIV_REQUIRES )

file(GLOB COMPONENT_SRCS
  LIST_DIRECTORIES false
  "*.h"
  "*.cpp"
  "*.c"
  "*.S"
)
set(COMPONENT_ADD_INCLUDEDIRS ".")

register_component()

```

valeros, 2020-05-20:
@wegunterjr I'm not an expert on ESP-IDF build system, but I'm not sure about an empty `COMPONENT_PRIV_REQUIRES`, besides it might be a good idea to try to specify all attributes in the `register_component` command.

wegunterjr, 2020-06-10:
I feel like I am the only one using platformio to do esp idf without arduino?   
so many things just don't work like they did prior to the update to esp idf 4.... 
thoughts?

wegunterjr, 2020-06-30:
so, it appears that there are a few changes to make to the cpp_utils because of the update to esp v4.

closed 2022-08-10 by stale[bot]