platformio/platform-espressif32 #342: upgrading to nwe netif code

DasVoelkel, 2020-05-07:
Since v4.1 updated the netif:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/tcpip_adapter_migration.html?highlight=tcpip_adapter_init

Is it possible to switch to the new system? As it is currently i can't find `esp_netif_init` in my program, only the old `tcpip_adapter_init` can I do this by hand or are there bigger things in the background that need to be changed ?

valeros, 2020-05-07:
Hi @DasVoelkel ! We support only the stable versions of frameworks. ESP-IDF `v4.1` hasn't been release yet. You can try to override the default package using the [platform_packages](https://docs.platformio.org/en/latest/projectconf/section_env_platform.html#platform-packages) option, but there is no guarantee that it'll work.

closed 2020-05-07 by valeros

DasVoelkel, 2020-05-07:
So I am guessing only when the next major will be released it will be supported by platformio?

valeros, 2020-05-07:
That's correct.

hassance3, 2020-05-15:
I hope this done soon!
Thank you.

DasVoelkel, 2020-05-16:
they come in yearls cycles