platformio/platform-espressif32 #344: ttgo-lora32-v2: pins_arduino.h: missing

peterus, 2020-05-09:
I am trying to compile my code for ttgo-lora32-v2, but it looks like there is an issue with a missing pins_arduino.h:

```
In file included from src/main.cpp:1:0:
/home/pbuchegger/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:185:26: fatal error: pins_arduino.h: No such file or directory
```

If i compile for ttgo-lora32-v1, it is working as expected but not for the ttgo-lora32-v2. If i take a look into `~/.platformio/packages/framework-arduinoespressif32/variants` i can see that the folder is missing. After a deinstall and install of the framework, there is no change.

jshank, 2020-05-22:
Confirming same issue for me.

valeros, 2020-05-25:
Looks like the variant for this board hasn't been released yet, so as a temporary workaround please use  `ttgo-lora32-v1` variant and the pin definitions provided by @jshank

jshank, 2020-06-02:
Looks like something went wrong with the [pull request](https://github.com/espressif/arduino-esp32/pull/3479/files) for this board has the updated `pins_arduino.h` but `~/.platformio/packages/framework-arduinoespressif32/variants/ttgo-lora32-v2/variants/pins_arduino.h` is still missing from the distribution as of 1.12.2.

jon7sky, 2020-06-11:
I just noticed the same thing.

jshank, 2020-06-12:
@valeros - Can you remove the board request label from this issue since the board has already been added, just not properly.

peterus, 2020-07-20:
is there something I can do to have this faster available? I would need the pin definition for my next project.

jshank, 2020-07-20:
@peterus just override the PIN definitions at the head of your file like I've shown above. I don't recall how many changes there were but you can reference [pins_arduino.h](https://github.com/espressif/arduino-esp32/pull/3479/files#diff-18434ff5954d81831c34077b810d29e5) for the correct definitions and redefine any pins you are using in your project.

> #define OLED_SDA 4
> #define OLED_SCL 15
> #define OLED_RST 16

peterus, 2020-07-20:
this is not a solution for me as i am sharing the project and a lot of people are depending on it. I could also overwrite the pin definitions directly in code, but this is also not nice if it can be done in the framework which is than working for everybody and every project in the future.

owenschoppe, 2020-08-30:
+1 to get this board added correctly. Is there another repo we should be commenting on?

SKRahulSK, 2020-09-17:
I am facing the same issue.
Please suggest if there is any update on this

DenHerrRing, 2020-09-24:
My Solution for that:
Open `[...]/.platformio\packages\framework-arduinoespressif32\variants`
Create Folder `ttgo-lora32-v2`
Create File `pins_arduino.h`
Insert this:
```
#ifndef Pins_Arduino_h
#define Pins_Arduino_h

#include <stdint.h>

#define EXTERNAL_NUM_INTERRUPTS 16
#define NUM_DIGITAL_PINS        40
#define NUM_ANALOG_INPUTS       16

#define analogInputToDigitalPin(p)  (((p)<20)?(esp32_adc2gpio[(p)]):-1)
#define digitalPinToInterrupt(p)    (((p)<40)?(p):-1)
#define digitalPinHasPWM(p)         (p < 34)

// I2C OLED Display works with SSD1306 driver
#define OLED_SDA    21
#define OLED_SCL    22
#define OLED_RST    16

// SPI LoRa Radio
#define LORA_SCK    5   // GPIO5 - SX1276 SCK
#define LORA_MISO   19  // GPIO19 - SX1276 MISO
#define LORA_MOSI   27  // GPIO27 - SX1276 MOSI
#define LORA_CS     18  // GPIO18 - SX1276 CS
#define LORA_RST    12  // GPIO14 - SX1276 RST
#define LORA_IRQ    26  // GPIO26 - SX1276 IRQ (interrupt request)

// SD card
#define SD_SCK  14
#define SD_MISO 2
#define SD_MOSI 15
#define SD_CS   13

static const uint8_t LED_BUILTIN =  22;
#define BUILTIN_LED  LED_BUILTIN    // backward compatibility

static const uint8_t KEY_BUILTIN =  0;

static const uint8_t TX =   1;
static const uint8_t RX =   3;

static const uint8_t SDA =  21;
static const uint8_t SCL =  22;

static const uint8_t SS =   18;
static const uint8_t MOSI = 27;
static const uint8_t MISO = 19;
static const uint8_t SCK =  5;

static const uint8_t A0 =   36;
static const uint8_t A1 =   37;
static const uint8_t A2 =   38;
static const uint8_t A3 =   39;
static const uint8_t A4 =   32;
static const uint8_t A5 =   33;
static const uint8_t A6 =   34;
static const uint8_t A7 =   35;
static const uint8_t A10 =  4;
static const uint8_t A11 =  0;
static const uint8_t A12 =  2;
static const uint8_t A13 =  15;
static const uint8_t A14 =  13;
static const uint8_t A15 =  12;
static const uint8_t A16 =  14;
static const uint8_t A17 =  27;
static const uint8_t A18 =  25;
static const uint8_t A19 =  26;

static const uint8_t T0 =   4;
static const uint8_t T1 =   0;
static const uint8_t T2 =   2;
static const uint8_t T3 =   15;
static const uint8_t T4 =   13;
static const uint8_t T5 =   12;
static const uint8_t T6 =   14;
static const uint8_t T7 =   27;
static const uint8_t T8 =   33;
static const uint8_t T9 =   32;

static const uint8_t DAC1 = 25;
static const uint8_t DAC2 = 26;

#endif /* Pins_Arduino_h */
```

Bert-Tec, 2021-02-05:
I'm having this issue too

dfsmith, 2021-02-11:
For my TTGO T3_V1.6 (supposedly a v2); the above with
`static const uint8_t LED_BUILTIN =  25;`

chrisco484, 2021-02-15:
I'm having this same issue with a Wemos D1 R2 (Uno-like board, not mini).

commit ddab9bbb (2021-02-24): Update Arduino core to v1.0.5

Resolve #298
Resolve #329
Resolve #344
Resolve #359
Resolve #378
Resolve #384
Resolve #389
Resolve #390
Resolve #414
Resolve #420
Resolve #436
Resolve #440
Resolve #493
closed 2021-02-24 by valeros