platformio/platform-espressif32 #347: How do you add a pairing code to a bluetooth service?

hzhh110, 2020-05-18:
How do you add a pairing code to a bluetooth service?

hzhh110, 2020-05-18:
`void initBLE(const char*name){
  DebugPrint("initBLE");
  DebugPrintln(name);
  BLEDevice::init(name); 
  // BLEDevice::init(name,"12345678");
  // Create the BLE Server
	BLEServer *pServer = BLEDevice::createServer();

	pServer->setCallbacks(new MyServerCallbacks());
	
	// Create the BLE Service
	BLEService *pService = pServer->createService(SERVICE_UUID);

	
	// Create a BLE Characteristic
	// pTXCharacteristic = pService->createCharacteristic(CHARACTERISTIC_UUID_TX, BLECharacteristic::PROPERTY_NOTIFY);
  pTXCharacteristic = pService->createCharacteristic(CHARACTERISTIC_UUID_TX, BLECharacteristic::PROPERTY_NOTIFY);
	                 
	pTXCharacteristic->addDescriptor(new BLE2902());
	
	BLECharacteristic *pRXCharacteristic = pService->createCharacteristic(CHARACTERISTIC_UUID_RX, BLECharacteristic::PROPERTY_WRITE);
	
	pRXCharacteristic->setCallbacks(new MyCallbacks());
	
	// Start the service
	pService->start();
	
	// Start advertising
	pServer->getAdvertising()->start();
 ]
}`

hzhh110, 2020-05-18:
How do you add a passkey to a bluetooth service?

kschoos, 2020-05-24:
Which framework are you using?

valeros, 2022-04-19:
Hi @romocitto88 ! This repository is not the right place to ask such generic questions not related to possible problems with the PlatformIO development platform for ESP32, please forward it to to https://gitter.im/espressif/arduino-esp32 or https://community.platformio.org/

closed 2022-04-19 by valeros