platformio/platform-espressif32 #348: Use PSRAM for malloc()

Avenitos, 2020-05-18:
Hello!

I use Platformio IDE +arduino-esp32 (1.12.1)

Use flags:

build_flags =
    -DBOARD_HAS_PSRAM
    -mfix-esp32-psram-cache-issue
PSRam are available for ps_malloc(), but not for malloc()

valeros, 2020-05-18:
Hi @Avenitos ! Does it work in Arduino IDE?

Avenitos, 2020-05-18:
I don't use the Arduino IDE, only Platformio vscode. I'll try to install it.

Avenitos, 2020-05-18:
I try in latest Arduino IDE.

`void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.print("Heap: ");
  Serial.print(ESP.getFreeHeap());
  Serial.print(" PsRam: ");  
  Serial.println(ESP.getFreePsram());  
  
  uint8_t *testMallocBuffer = (uint8_t*)malloc(11000);
  if(testMallocBuffer == NULL) Serial.println("malloc failed");
  
  delay(5000);
}`

`Heap: 353420 PsRam: 4194252
Heap: 342404 PsRam: 4194252
Heap: 331388 PsRam: 4194252
Heap: 320372 PsRam: 4194252
Heap: 309356 PsRam: 4194252
Heap: 298340 PsRam: 4194252
Heap: 287324 PsRam: 4194252
Heap: 276308 PsRam: 4194252
Heap: 265292 PsRam: 4194252
`

valeros, 2020-05-18:
Thanks!
What is the output when you compile this code using PlatformIO?

Avenitos, 2020-05-18:
`> Executing task in folder test_psram: C:\.platformio\penv\Scripts\platformio.exe run <

Processing esp-wrover-kit (platform: espressif32; board: esp-wrover-kit; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp-wrover-kit.html
PLATFORM: Espressif 32 1.12.1 > Espressif ESP-WROVER-KIT
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (ftdi) On-board (ftdi) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.200129 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\esp-wrover-kit\src\main.cpp.o
Generating partitions .pio\build\esp-wrover-kit\partitions.bin
Archiving .pio\build\esp-wrover-kit\libFrameworkArduinoVariant.a
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\Esp.cpp.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\FunctionalInterrupt.cpp.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\IPv6Address.cpp.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\MD5Builder.cpp.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\Print.cpp.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\StreamString.cpp.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\WString.cpp.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\base64.cpp.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\cbuf.cpp.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-adc.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-bt.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-cpu.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-dac.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-gpio.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-i2c.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-ledc.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-matrix.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-misc.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-psram.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-rmt.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-sigmadelta.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-spi.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-time.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-timer.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-touch.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-uart.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\libb64\cdecode.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\libb64\cencode.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\main.cpp.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\stdlib_noniso.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\esp-wrover-kit\libFrameworkArduino.a
Linking .pio\build\esp-wrover-kit\firmware.elf
Building .pio\build\esp-wrover-kit\firmware.bin
Retrieving maximum program size .pio\build\esp-wrover-kit\firmware.elf
Checking size .pio\build\esp-wrover-kit\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   4.7% (used 15444 bytes from 327680 bytes)
Flash: [==        ]  16.6% (used 217765 bytes from 1310720 bytes)
esptool.py v2.6
======================================================== [SUCCESS] Took 8.95 seconds ========================================================

Terminal will be reused by tasks, press any key to close it.`

valeros, 2020-05-18:
I meant the output from the board.

Avenitos, 2020-05-18:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Heap: 353356 PsRam: 4194252
Heap: 342340 PsRam: 4194252
Heap: 331324 PsRam: 4194252
Heap: 320308 PsRam: 4194252
```

valeros, 2020-05-18:
So it looks it should be enabled explicitly
https://docs.espressif.com/projects/esp-idf/en/v3.3.2/api-guides/external-ram.html#provide-external-ram-via-malloc
I'm closing the issue as it's not PlatformIO specific.

closed 2020-05-18 by valeros