platformio/platform-espressif32 #349: Add support for ESP32-S2 Dev Boards

THATDONFC, 2020-05-20:
What kind of issue is this?

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

------------------------------------------------------------------
### Additional info

I didn't see any information regarding the new ESP32-S2 Dev Boards from Espressif.

Specifically the [ESP32-S2-Saola-1](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-saola-1-v1.2.html) based on the [ESP32-S2-WROVER](https://www.espressif.com/sites/default/files/documentation/esp32-s2-wrover_esp32-s2-wrover-i_datasheet_en.pdf) and [ESP32-S2-WROOM](https://www.espressif.com/sites/default/files/documentation/esp32-s2-wroom_esp32-s2-wroom-i_datasheet_en.pdf)
and ESP32-S2-Kaluga-1 based on the [ESP32-S2-WROVER](https://www.espressif.com/sites/default/files/documentation/esp32-s2-wrover_esp32-s2-wrover-i_datasheet_en.pdf)

Will we see these boards soon?

valeros, 2020-05-20:
Hi @THATDONFC ! Thanks for the request. ESP32-S2-based boards will be added as soon a stable version of Arduino or ESP-IDF with support for this MCU is released.

jcwren, 2020-06-10:
What constitutes a stable version? I've got some S2 modules coming in and I'd love to be able to develop for those under platformio.

tschnoelzer, 2020-07-02:
Any news on this guys? My ESP32-S2-Saola-1M just arrived ...

TuringMachinegun, 2020-07-16:
Seems like there's no stable release yet from Arduino / ESP-IDF supporting these boards.

For ESP-IDF, currently only version 4.2+ [supports them](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-saola-1-v1.2.html#software-setup), and right now only v4.0 [has a stable release](https://github.com/espressif/esp-idf/releases).

Same for Arduino support, still [on a branch](https://github.com/espressif/arduino-esp32/tree/esp32s2) which has not been merged to master yet.

knifter, 2020-07-16:
Just a thought: Does this actually need to be developed in this 'platform'? I mean, considering all the 'if's required in the preliminary  support commit it could be considered to fork this repos to a platform-espressif32**s2**? On the other hand, it would require updating/merging to that as well with every IDF update.

askpatrickw, 2020-09-18:
Seems like this is in release 2.0.0 as a Beta.
https://github.com/platformio/platform-espressif32/releases/tag/v2.0.0

FrGe2016, 2020-11-14:
I just maked the acquisition of a ESP32-S2-SOLA, it seems that it is still not availlable with Platform IO is there any expected date.

P.S i have been able to import the s2 branch for the Arduino-IDE but it is not my best development environment.

cd espressif && \
--
git clone -b esp32s2 https://github.com/espressif/arduino-esp32.git esp32 && \
cd esp32 && \
git submodule update --init --recursive && \
cd tools && \
python3 get.py

mannro, 2020-11-16:
same question here

nivoc, 2020-12-03:
As @GiovanniCmpanerpointed out in Issue #432. If you are ok with ArduinoLib you can use this settings in your plattform.ini to use Arduinos 4.2 Release with S2 Support.

```
[env:esp32doit-devkit-v1]
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.2
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
board_build.mcu = esp32s2
```

tavdog, 2020-12-10:
I tried this with the UM FeatherS2 ESP32S2 board and it compiles and appears to upload but it doesn't activate the serial port like it does when programmed in the Arduino IDE.  If there was just a way to install the the ESP32S2 branch from espressif into pio   https://github.com/espressif/arduino-esp32/tree/esp32s2

septatrix, 2020-12-10:
ESP-IDF v4.2 was released three day ago (https://github.com/espressif/esp-idf/releases/tag/v4.2). That release also brings support for the ESP32-S2.

Maybe someone with a better knowledge of how this repository works could submit a PR which updates the bundled framework version.

jcwren, 2020-12-10:
And yet still no support for USB OTG or WiFi Time-Of-Flight :(

septatrix, 2020-12-10:
> And yet still no support for USB OTG or WiFi Time-Of-Flight :(

A quick search in the linked repo resulted in this: https://github.com/espressif/esp-idf/issues/5059#issuecomment-735050296. Time of flight is not yet supported but planned for the next release. Looking in the documentation and the forum post referenced in the comment shows that ToF, USB Host and some other stuff is planned for v4.3.

Mark-81, 2021-01-20:
I read this issue and the others but I don't understand if right now is possible to write code for ESP32-S2 with the Arduino framework under PlatformIO VS.

fracool, 2021-01-20:
Hi mark! Yes it is possible, it's mentioned above too, but here is an example platformio.ini you could use to achieve this:
`[env:esp32dev]

platform = espressif32

board = esp32dev

framework = arduino

monitor_speed = 115200

lib_deps =`

sblantipodi, 2021-01-20:
> Hi mark! Yes it is possible, it's mentioned above too, but here is an example platformio.ini you could use to achieve this:
> [env:esp32dev]
> platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.2
> platform = espressif32
> board = esp32dev
> framework = arduino
> board_build.mcu = esp32s2
> monitor_speed = 115200

but it is fully supported?
what about the wifi features/libraries?

are they compatible with the ESP32 one?

fracool, 2021-01-20:
I'm using the  espressif ESP32-S2-Saola-1and so far everything works.

gcormier, 2021-01-20:
I assume it will also work for the new ESP32-S2-MINI since it's also just
an S2 supported by ESPIDF?

fracool, 2021-01-20:
yeah I think any s2 would be fine with these settings

sblantipodi, 2021-01-21:
are you able to use wifi?
I'm getting this error

```
Compiling .pio\build\glowwormluciferinfull_esp32_s2\lib346\WiFi\WiFiGeneric.cpp.o
xtensa-esp32-elf-g++: error: unrecognized command line option '-mfix-esp32-psram-cache-strategy=memw'
```

fracool, 2021-01-21:
It looks like your build config is wrong. are you using the same platformio.ini file I mentioned? It should be trying to use xtensa-esp32s2-elf-g++, did you specify the platform package?

platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.2

I did have similar frustrations while I was trying and it was due to me having the wrong micro processor set. 

board_build.mcu = esp32s2

I am able to use everything I've tried so far with no probs - WiFi works, espnow, espmesh etc 

The only thing I did to get it working was what I posted in my last comment!

sblantipodi, 2021-01-22:
> It looks like your build config is wrong. are you using the same platformio.ini file I mentioned? It should be trying to use xtensa-esp32s2-elf-g++, did you specify the platform package? [...]

yes using this:
```
[env:esp32dev]
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.2
platform = espressif32
board = esp32dev
framework = arduino
board_build.mcu = esp32s2
monitor_speed = 115200
```

I have also other errors:
/cores/esp32/Arduino.h:33:10: fatal error: freertos/FreeRTOS.h: No such file or directory
 #include "freertos/FreeRTOS.h"

valeros, 2021-01-25:
Hi guys, I've pushed updates for ESP-IDF v4.2, but at the moment without support for mixed Arduino, ESP-IDF projects. In case someone wants to try it out, please retest with the upstream version of the platform:
```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
framework = espidf
board = esp32dev
monitor_speed = 115200
```

closed 2021-01-25 by valeros

kvn1351, 2021-04-06:
Amazing, I've been looking forward to this.