platformio/platform-espressif32 #353: Esp32 flash size in menuconfig doesn’t work

romocitto88, 2020-05-27:
Hi,
I want to set flash size of my project from 4MB in 8MB, but it doesn’t change!!

This is the Setup in menuconfig:

[image]

This is the error:

[image]

Can someone help me?

brad-colbert, 2020-12-02:
How did you get the 4MB?  Mine is still stuck at 1MB.

valeros, 2021-01-06:
Have you tried to override the flash size directly in the `platformio.ini` file, e.g.:

```ini
board_upload.maximum_size = 8388608 
```

romocitto88, 2021-01-20:
In the end I created a custom board file, with the flash size for my esp module.
I will close this issue. 

Thank you.

closed 2021-01-20 by romocitto88