platformio/platform-espressif32 #355: PIO remote upload fails for espidf framework

nathanjel, 2020-05-30:
When trying to do `pio remote -e esp32dev -t upload` I get the following error.
Otherwise, the same project builds and uploads well when used locally.
I checked that the project binary files are copied to the remote machine and the project folder exists. There is no "src" folder, I assume the espidf.py build script needs updates to work in remote mode after recent CMake related updates.

```Uploading firmware remotely
Processing esp32dev (platform: espressif32; framework: espidf; board: esp32dev; monitor_speed: 115200; lib_deps: ElectronSquareCore, ElectronSquareAPI; lib_extra_dirs: /home/nathan/work/e2/libs)
--------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.12.2 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf 3.40001.200521 (4.0.1) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - tool-mkspiffs 2.230.0 (2.30)
Reading CMake configuration...

FileNotFoundError: [WinError 3] System nie mo�e odnale�� okre�lonej �cie�ki: 'C:\\Users\\marci\\.platformio\\remote\\projects\\master-459ccf860f5e72a8a9dbe0ec13e21f6e17ecf594\\src':
  File "C:\Users\marci\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 168:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\marci\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\marci\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\marci\.platformio\platforms\espressif32\builder\main.py", line 213:
    target_elf = env.BuildProgram()
  File "C:\Users\marci\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\marci\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\marci\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\marci\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 118:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\marci\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\marci\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 330:
    SConscript(env.GetFrameworkScript(f), exports="env")
  File "C:\Users\marci\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 660:
    return method(*args, **kw)
  File "C:\Users\marci\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\marci\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\marci\.platformio\platforms\espressif32\builder\frameworks\espidf.py", line 856:
    project_codemodel = get_cmake_code_model(
  File "C:\Users\marci\.platformio\platforms\espressif32\builder\frameworks\espidf.py", line 208:
    create_default_project_files()
  File "C:\Users\marci\.platformio\platforms\espressif32\builder\frameworks\espidf.py", line 178:
    if not listdir(join(env.subst("$PROJECT_SRC_DIR"))):
========================== [FAILED] Took 0.58 seconds ==========================
Environment    Status    Duration
-------------  --------  ------------
esp32dev       FAILED    00:00:00.582
stm32          IGNORED
==================== 1 failed, 0 succeeded in 00:00:00.582 ====================
```

valeros, 2020-06-01:
Many thanks for the report! I've seen your PR, but the build script for ESP-IDF is not the right place to fix this bug. Please retest with the upstream version of the platform https://docs.platformio.org/en/latest/platforms/espressif32.html#stable-and-upstream-versions

closed 2020-06-01 by valeros

nathanjel, 2020-06-01:
Thank You. Tested with upstream and it works. Indeed my understanding of PIO build system and Cons is somewhat limited, learning by doing. Your solution is obviously better, not only well placed but seems to run a bit faster as the unnecessary actions are not executed.

nathanjel, 2020-06-02:
When this gets merged into master?

commit 68874863 (2020-06-01): Fix unwanted build step for nobuild target // Resolve #355, resolve #356