platformio/platform-espressif32 #362: Build fails if library provided as ZIP file

nathanjel, 2020-06-09:
It seems that ESP-IDF builder calls CMake before installing external libraries from .zip files. First run of `platformio ci` fails and second run (assuming the build folder is kept) goes OK. Log output suggests the sequence is invalid - first CMake run and only then library installation. I've seen this before with `.zip` libraries and `platformio run` but then it was so easy to restart. When using CI this is a real pain. Happy to work on resolution, but would need some suggestion where to start.

Below steps to reproduce and platformio.ini content.

```
[env:esp32dev-sample]
platform = https://github.com/platformio/platform-espressif32.git
framework = espidf
board = esp32dev
monitor_speed = 115200
lib_deps =
  http://[...secret...]/ElectronSquareHALESP32.zip
```

---

`rm -rf /home/nathan/tmp`

---

`pio ci --project-conf=ci/platformio.ini --lib=. examples/sample.cpp --build-dir=/home/nathan/tmp --keep-build-dir`

---

```
The next files/directories have been created in /home/nathan/tmp
include - Put project header files here
lib - Put here project specific (private) libraries
src - Put project source files here
platformio.ini - Project Configuration File

Project has been successfully updated! Useful commands:
`pio run` - process/build project from the current directory
`pio run --target upload` or `pio run -t upload` - upload firmware to a target
`pio run --target clean` - clean project (remove compiled files)
`pio run --help` - additional information
Processing esp32dev-sample (platform: https://github.com/platformio/platform-espressif32.git; framework: espidf; board: esp32dev)
-------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.12.2 #6887486 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf 3.40001.200521 (4.0.1) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - tool-ninja 1.7.1 
 - toolchain-esp32ulp 1.22851.190618 (2.28.51) 
 - toolchain-xtensa32 2.80200.200226 (8.2.0)
Reading CMake configuration...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing ElectronSquareHALESP32
Using cache: /home/nathan/.platformio/.cache/34/c0514419c48f945ba76c27ecbb40f734
ElectronSquareHALESP32 @ 0.9 has been successfully installed!
Found 2 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ElectronSquareHALESP32> 0.9
|-- <ElectronSquareButton> 0.9
|   |-- <ElectronSquareHALESP32> 0.9
Building in release mode
Compiling .pio/build/esp32dev-sample/esp-idf/src/sample.cpp.o
Generating LD script .pio/build/esp32dev-sample/esp32_out.ld
In file included from src/sample.cpp:1:
lib/ElectronSquareButton/include/BUTTON.h:18:10: fatal error: GPIO.h: No such file or directory

**************************************************************
* Looking for GPIO.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:GPIO.h"
* Web  > https://platformio.org/lib/search?query=header:GPIO.h
*
**************************************************************

 #include "GPIO.h"
          ^~~~~~~~
compilation terminated.
Generating partitions .pio/build/esp32dev-sample/partitions.bin
Compiling .pio/build/esp32dev-sample/esp-idf/app_trace/app_trace.c.o
Compiling .pio/build/esp32dev-sample/esp-idf/app_trace/app_trace_util.c.o
Compiling .pio/build/esp32dev-sample/esp-idf/app_trace/host_file_io.c.o
Compiling .pio/build/esp32dev-sample/esp-idf/app_trace/gcov/gcov_rtio.c.o
Compiling .pio/build/esp32dev-sample/esp-idf/app_update/esp_ota_ops.c.o
Compiling .pio/build/esp32dev-sample/esp-idf/app_update/esp_app_desc.c.o
Compiling .pio/build/esp32dev-sample/esp-idf/asio/asio/asio/src/asio.cpp.o
Compiling .pio/build/esp32dev-sample/esp-idf/bootloader_support/src/bootloader_clock.c.o
*** [.pio/build/esp32dev-sample/esp-idf/src/sample.cpp.o] Error 1
Compiling .pio/build/esp32dev-sample/esp-idf/bootloader_support/src/bootloader_common.c.o
================================================= [FAILED] Took 7.14 seconds 
```

---

This just avoids error message, but leaves the build dir otherwise intact
`rm -rf /home/nathan/tmp/lib/ElectronSquareButton`

---

`pio ci --project-conf=ci/platformio.ini --lib=. examples/sample.cpp --build-dir=/home/nathan/tmp --keep-build-dir`

---

```
The next files/directories have been created in /home/nathan/tmp
include - Put project header files here
lib - Put here project specific (private) libraries
src - Put project source files here
platformio.ini - Project Configuration File

Project has been successfully updated! Useful commands:
`pio run` - process/build project from the current directory
`pio run --target upload` or `pio run -t upload` - upload firmware to a target
`pio run --target clean` - clean project (remove compiled files)
`pio run --help` - additional information
Processing esp32dev-sample (platform: https://github.com/platformio/platform-espressif32.git; framework: espidf; board: esp32dev)
-------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.12.2 #6887486 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf 3.40001.200521 (4.0.1) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - tool-ninja 1.7.1 
 - toolchain-esp32ulp 1.22851.190618 (2.28.51) 
 - toolchain-xtensa32 2.80200.200226 (8.2.0)
Reading CMake configuration...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 2 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ElectronSquareHALESP32> 0.9
|-- <ElectronSquareButton> 0.9
|   |-- <ElectronSquareHALESP32> 0.9
Building in release mode
Compiling .pio/build/esp32dev-sample/esp-idf/src/sample.cpp.o
[...]
Building .pio/build/esp32dev-sample/firmware.bin
Checking size .pio/build/esp32dev-sample/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   3.9% (used 12644 bytes from 327680 bytes)
Flash: [==        ]  15.7% (used 164109 bytes from 1048576 bytes)
esptool.py v2.6
================================================ [SUCCESS] Took 18.67 seconds
```

nathanjel, 2020-06-09:
There is an easy workaround for the build process in `run` mode - call `platformio lib install` before `platformio run` allows to avoid this issue. No idea how to do a similar thing in CI scenario.

valeros, 2020-06-11:
Hi @nathanjel ! So, It only fails with libraries added as ZIP files? Is there a Github project I can fork and try to reproduce the issue?

nathanjel, 2020-06-12:
Hi @valeros ! Unfortunately, it's a general issue, not only with ZIP files but with libs declared in `platformio.ini` for the `espidf` framework. I tested with a PIO library and with a git repo. I have a vested interest in getting this fixed, so I created a dedicated sample to reproduce the issue. When trying `pio ci` the issue is the same, but there is no way to work it around with `pio lib install`. Please note that with some hint, I'd be happy to work on resolving this.

Clone: https://github.com/nathanjel/id0-pio-proj.git
Lastet version of pio (`pio upgrade`, `pio update`)

Test 1
`rm -rf .pio`
`pio run` - the first run will fail, reporting a missing header file `storage/memory.h` from the `emdb` library
```
Environment                Status    Duration
-------------------------  --------  ------------
esp32dev-id0-pio-test-git  FAILED    00:00:39.797
esp32dev-id0-pio-test-lib  FAILED    00:00:07.944
```
`pio run` - the second run will complete successfully
```
Environment                Status    Duration
-------------------------  --------  ------------
esp32dev-id0-pio-test-git  SUCCESS   00:00:19.404
esp32dev-id0-pio-test-lib  SUCCESS   00:00:20.602
```

Test 2
`rm -rf .pio`
`pio lib install`
`pio run` - will complete successfuly

nathanjel, 2020-06-12:
Hi again @valeros ! It's getting weird now - please see also the `https://github.com/nathanjel/id0-pio-proj.git` project on branch `issue/platformio-362-uncertain`. 

The problem occurs with `emdb` library but not with `jsmn`. The latter has the `.h` file in library root directory, while `emdb` has include files in `include` - as my original issue. Hope this observation helps.

The fix I proposed in the merge requests resolves the problem, but does not explain it or it's root cause.

valeros, 2020-09-02:
Hi @nathanjel ! Sorry for the late reply, should be resolved in the dev branch with ESP-IDF v4.1, please retest with the upstream version:
```
[env:upstream_develop]
platform = https://github.com/platformio/platform-espressif32.git
framework = esp-idf
board = ...
```

closed 2020-12-28 by nathanjel