platformio/platform-espressif32 #365: Unable to run menuconfig with stable release or upstream

wegunterjr, 2020-06-12:
I am just testing the ability to run menuconfig, but 
OS - windows
Platformio espressif32 1.12.3

I am getting this:
`PackageManager: Installing toolchain-xtensa32 @ ~2.80200.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
toolchain-xtensa32 @ 2.80200.200226 has been successfully installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp-wrover-kit.html
PLATFORM: Espressif 32 1.12.3 #6d85d3b > Espressif ESP-WROVER-KIT
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (ftdi) On-board (ftdi) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf 3.40001.200521 (4.0.1) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - tool-ninja 1.7.1
 - toolchain-esp32ulp 1.22851.190618 (2.28.51)
 - toolchain-xtensa32 2.80200.200226 (8.2.0)
Reading CMake configuration...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Running menuconfig...
cc -c  -DCURSES_LOC="<curses.h>" -DLOCALE -MMD -MP -I "." -I "/root/.platformio/packages/framework-espidf/tools/kconfig"  /root/.platformio/packages/framework-espidf/tools/kconfig/mconf.c -o mconf.o
<command-line>:0:12: fatal error: curses.h: No such file or directory
compilation terminated.
/root/.platformio/packages/framework-espidf/tools/kconfig/Makefile:173: recipe for target 'mconf.o' failed
make: *** [mconf.o] Error 1
ninja: build stopped: subcommand failed.
Error: Couldn't execute 'menuconfig' target.
scons: *** [menuconfig] Explicit exit, status 1
root@Alien-Magic:/mnt/c/Users/wegun/Documents/PlatformIO/Projects/test_esp32#   ============== [FAILED] Took 46.79 seconds ============================================================================================== 
`

valeros, 2020-06-13:
@wegunterjr Several additional packages need to be installed on Linux-based systems:
```
libncurses5-dev flex bison
```
https://docs.platformio.org/en/latest/frameworks/espidf.html#configuration-for-4-0

closed 2020-06-13 by valeros

wegunterjr, 2020-06-13:
Right ..I'm running this on Windows.  I thought I was running this in a
Platformio terminal in vscode (not wsl or cmd)   I'll verify but I'll try
uninstalling, deleting Platformio folder, and then reinstall Platformio as
well.

Thanks!

wegunterjr, 2020-06-13:
@valeros hmmm...using two different windows machines and getting two different results. one windows machines (seems to be using anaconda python env) works fine, but the other windows machine doesn't have that and isn't working...
Doesn't all the right software install when I install platformio?

wegunterjr, 2020-06-13:
aha!   It is because I have WSL on Windows and it was default to WSL instead of CMD... so, i changeed the default to CMD and then it  yelled at me for having a small screen, so i fixed that...
your comment about linux got me thinking and I realized what it was.

`Your display is too small to run Menuconfig!ts/Projects/connect-esp32/sdkconfig.defaults...
It must be at least 19 lines by 80 columns.
Batch file failed at line 4 with errorcode 1`