platformio/platform-espressif32 #367: Compile error: Adafruit GFX libraries won't find Wire.h / SPI.h

xsrf, 2020-06-22:
Hi,
whenever I start a new project in Platform IO for ESP8266 or ESP32 using Adafruit GFX Library I get a bunch of compile errors.
The libraries using Wire.h / SPI.h won't find them, but they're clearly there, I can use them in my main code.

_Sorry if this is the wrong place to report this issue, but I really don't know if this is an issue with Adafruit GFX, ESP8266/ESP32 Platform in Plaform IO, Platform IO VSCode IDE, Platform IO Core or VSCode... I don't want to spam all repositories with this. Already opened an Issue here: https://github.com/adafruit/Adafruit-GFX-Library/issues/307_

main.cpp
```
#include <Arduino.h>
#include <Wire.h>

void setup() {
  Wire.begin();
}

void loop() {
  // put your main code here, to run repeatedly:
}
```
platformio.ini for ESP32:
```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
lib_deps = Adafruit GFX Library
```
<details>
<summary>Compiler output</summary>

```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 1.12.1 > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing id=13
Using cache: C:\Users\***\.platformio\.cache\cb\eb4a83205a6bf7664c071bda082bc1cb
Adafruit GFX Library @ 1.9.0 has been successfully installed!
Installing dependencies
Looking for Adafruit BusIO library in registry
Found: https://platformio.org/lib/show/6214/Adafruit BusIO
LibraryManager: Installing id=6214
Using cache: C:\Users\***\.platformio\.cache\87\51e35a894cbff55174e7997f44b7d387
Adafruit BusIO @ 1.3.2 has been successfully installed!
Found 28 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Adafruit GFX Library> 1.9.0
|   |-- <Adafruit BusIO> 1.3.2
Building in release mode
Compiling .pio\build\esp32doit-devkit-v1\src\main.cpp.o
Generating partitions .pio\build\esp32doit-devkit-v1\partitions.bin
Compiling .pio\build\esp32doit-devkit-v1\lib1a3\Adafruit BusIO_ID6214\Adafruit_BusIO_Register.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib1a3\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib1a3\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp.o
In file included from .pio\libdeps\esp32doit-devkit-v1\Adafruit BusIO_ID6214/Adafruit_BusIO_Register.h:1:0,
                 from .pio\libdeps\esp32doit-devkit-v1\Adafruit BusIO_ID6214\Adafruit_BusIO_Register.cpp:1:
.pio\libdeps\esp32doit-devkit-v1\Adafruit BusIO_ID6214/Adafruit_I2CDevice.h:1:18: fatal error: Wire.h: No such file or directory

**************************************************************
* Looking for Wire.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Wire.h"
[... 155 lines cut ...]

 #include <SPI.h>
                 ^
compilation terminated.
*** [.pio\build\esp\lib166\Adafruit GFX Library_ID13\Adafruit_MonoOLED.cpp.o] Error 1
In file included from .pio\libdeps\esp\Adafruit GFX Library_ID13\Adafruit_SPITFT.cpp:36:0:
.pio\libdeps\esp\Adafruit GFX Library_ID13\Adafruit_SPITFT.h:26:17: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://platformio.org/lib/search?query=header:SPI.h
*
*************************************************************

 #include <SPI.h>
                 ^
compilation terminated.
*** [.pio\build\esp\lib166\Adafruit GFX Library_ID13\Adafruit_SPITFT.cpp.o] Error 1
================================================================================================== [FAILED] Took 2.46 seconds ==================================================================================================
The terminal process terminated with exit code: 1
```

</details>

valeros, 2020-06-22:
Looks like the problem is similar to https://github.com/adafruit/Adafruit-GFX-Library/issues/298#issuecomment-636046011

xsrf, 2020-06-22:
@valeros 
I don't see how the `lib_ignore` approach by @ivankravets would help. The issue is that
```
.pio\libdeps\esp32doit-devkit-v1\Adafruit GFX Library_ID13\Adafruit_MonoOLED.cpp
.pio\libdeps\esp32doit-devkit-v1\Adafruit GFX Library_ID13\Adafruit_MonoOLED.h
.pio\libdeps\esp32doit-devkit-v1\Adafruit GFX Library_ID13\Adafruit_SPITFT.cpp
.pio\libdeps\esp32doit-devkit-v1\Adafruit GFX Library_ID13\Adafruit_SPITFT.h
.pio\libdeps\esp32doit-devkit-v1\Adafruit GFX Library_ID13\Adafruit_SPITFT_Macros.h
```
do not compile because of missing Wire.h / SPI.h. These are part of the core Adafruit GFX library, I can't ignore them...

valeros, 2020-06-22:
You don't need to ignore them, you need to explicitly specify them as dependencies:
```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
lib_deps =
    Adafruit GFX Library
    Wire
    SPI
```

closed 2020-06-22 by valeros

xsrf, 2020-06-22:
okay, so PlaformIO will detect that Wire/SPI etc is needed automatically if it is used in the main program, but not when used by libraries?
Adding both headers to my main program works... But it seems odd, because I don't use Wire/SPI anywhere in my code.
But it works now, thx!

valeros, 2020-06-22:
> okay, so PlaformIO will detect that Wire/SPI etc is needed automatically if it is used in the main program, but not when used by libraries?

That's true, that dependencies are probably hidden somewhere in the sources of external libraries, so the LDF is not able to detect them in default mode. You can control this behaviour by using other LDF modes https://docs.platformio.org/en/latest/librarymanager/ldf.html if .

shan-mcarthur, 2023-03-20:
So what does Adafruit need to do differently to have the dependencies automatically picked up?