platformio/platform-espressif32 #369: WifiMulti.h lib

Tony763, 2020-06-24:
Hello, I trying to convert [ESP32Radio](https://github.com/Edzelf/ESP32-Radio) project to use with PlatformIO. I created a new project with `platformio.ini` containing:

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
platform_packages =
    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
lib_ldf_mode = off
lib_deps =
    Wire
    SPI
    Adafruit GFX Library
    Adafruit ILI9341
    1744        ;ESP_VS1053_Library
```

I have a problem with dependency: `WiFiMulti.h`  In Arduino IDE it is included in platform libraries. [See](https://github.com/espressif/arduino-esp32/tree/5cfff190e9bc6edd7c9dfc31f8af50957c863872/libraries/WiFi/src)

valeros, 2020-06-24:
You disabled LDF, that's why you need to manually specify the `WiFi` library as a dependency in your `lib_deps` list.

closed 2020-06-24 by valeros

Tony763, 2020-06-25:
Did not help.
I run it with `deep+`
Get:
```
Dependency Graph
|-- <SimpleBLE> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.2.7
|   |   |-- <SPI> 1.0
|-- <Wire> 1.0.1
|-- <SPI> 1.0
|-- <FS> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <WiFi> 1.2.7
|   |   |-- <SPI> 1.0
|   |-- <Update> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.2.7
|   |   |   |-- <SPI> 1.0
|-- <Update> 1.0
|-- <Adafruit GFX Library> 1.9.0
|   |-- <Adafruit BusIO> 1.3.2
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <Adafruit ILI9341> 1.5.5
|   |-- <Adafruit GFX Library> 1.9.0
|   |   |-- <Adafruit BusIO> 1.3.2
|   |   |   |-- <Wire> 1.0.1
|   |   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <Adafruit STMPE610> 1.1.2
|   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0.1
|   |-- <Adafruit TouchScreen> 1.1.0
|   |-- <SPI> 1.0
|-- <ESP_VS1053_Library> 1.1.2
|   |-- <SPI> 1.0
|-- <WiFi> 1.0
|   |-- <SimpleBLE> 1.0
|-- <SD(esp32)> 1.0.5
|   |-- <FS> 1.0
|   |-- <SPI> 1.0
```

When I specify all found libraries I still get error.
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
platform_packages =
    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
lib_ldf_mode = off
lib_deps =
    SimpleBLE
    ESPmDNS
    Wifi    
    Wire
    SPI
    FS
    ArduinoOTA
    Update
    Adafruit GFX Library
    Adafruit ILI9341
    Adafruit STMPE610
    Adafruit TouchScreen
    Adafruit BusIO
    SD(esp32)
    1744        ;ESP_VS1053_Library
```

```
Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.12.4 #c4220ce > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 b92c58d
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Looking for Wifi library in registry
Found: https://platformio.org/lib/show/870/WiFi
LibraryManager: Installing id=870
WiFi @ 1.2.7 is already installed
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SimpleBLE> 1.0
|-- <ESPmDNS> 1.0
|-- <Wire> 1.0.1
|-- <SPI> 1.0
|-- <FS> 1.0
|-- <ArduinoOTA> 1.0
|-- <Update> 1.0
|-- <Adafruit GFX Library> 1.9.0
|   |-- <Adafruit BusIO> 1.3.2
|-- <Adafruit ILI9341> 1.5.5
|   |-- <Adafruit GFX Library> 1.9.0
|   |   |-- <Adafruit BusIO> 1.3.2
|   |-- <Adafruit STMPE610> 1.1.2
|   |-- <Adafruit TouchScreen> 1.1.0
|-- <Adafruit STMPE610> 1.1.2
|-- <Adafruit TouchScreen> 1.1.0
|-- <Adafruit BusIO> 1.3.2
|-- <SD(esp32)> 1.0.5
|-- <ESP_VS1053_Library> 1.1.2
Building in release mode
Compiling .pio\build\esp32dev\src\main.cpp.o
Compiling .pio\build\esp32dev\src\src\pubsubclient-master\PubSubClient.cpp.o
Generating partitions .pio\build\esp32dev\partitions.bin
Compiling .pio\build\esp32dev\libc61\SimpleBLE\SimpleBLE.cpp.o
Compiling .pio\build\esp32dev\lib216\ESPmDNS\ESPmDNS.cpp.o
Compiling .pio\build\esp32dev\libbfb\Wire\Wire.cpp.o
Compiling .pio\build\esp32dev\libc87\SPI\SPI.cpp.o
src\main.cpp:177:23: fatal error: WiFiMulti.h: No such file or directory

*******************************************************************
* Looking for WiFiMulti.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFiMulti.h"
* Web  > https://platformio.org/lib/search?query=header:WiFiMulti.h
*
*******************************************************************

compilation terminated.
Compiling .pio\build\esp32dev\lib640\FS\FS.cpp.o
Compiling .pio\build\esp32dev\lib640\FS\vfs_api.cpp.o
C:\Users\Tony\.platformio\packages\framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6\libraries\ESPmDNS\src\ESPmDNS.cpp:42:18: fatal error: WiFi.h: No such file or directory

**************************************************************
* Looking for WiFi.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFi.h"
* Web  > https://platformio.org/lib/search?query=header:WiFi.h
*
**************************************************************

compilation terminated.
Compiling .pio\build\esp32dev\libe5d\ArduinoOTA\ArduinoOTA.cpp.o
Compiling .pio\build\esp32dev\lib73d\Update\Updater.cpp.o
C:\Users\Tony\.platformio\packages\framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6\libraries\ArduinoOTA\src\ArduinoOTA.cpp:5:21: fatal error: WiFiUdp.h: No such file or directory

*****************************************************************
* Looking for WiFiUdp.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFiUdp.h"
* Web  > https://platformio.org/lib/search?query=header:WiFiUdp.h
*
*****************************************************************

compilation terminated.
Compiling .pio\build\esp32dev\lib279\Adafruit BusIO_ID6214\Adafruit_BusIO_Register.cpp.o
*** [.pio\build\esp32dev\src\main.cpp.o] Error 1
*** [.pio\build\esp32dev\lib216\ESPmDNS\ESPmDNS.cpp.o] Error 1
*** [.pio\build\esp32dev\libe5d\ArduinoOTA\ArduinoOTA.cpp.o] Error 1
===================================================================================================== [FAILED] Took 3.91 seconds =====================================================================================================The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

valeros, 2020-06-25:
> I run it with deep+

What is the error in that case?

> When I specify all found libraries I still get error.

Library names are case-sensitive, so `Wifi` should be `WiFi`

Tony763, 2020-06-25:
> What is the error in that case?
```
src\main.cpp: In function 'void update_software(const char*, const char*, const char*)':
src\main.cpp:2625:40: error: 'scan_content_length' was not declared in this scope
     scan_content_length ( line.c_str() ) ;                      // Scan for content_length
                                        ^
.pio\libdeps\esp32dev\WiFi_ID870\src\utility\wifi_drv.cpp: In static member function 'static uint8_t WiFiDrv::getEncTypeNetowrks(uint8_t)':
.pio\libdeps\esp32dev\WiFi_ID870\src\utility\wifi_drv.cpp:451:10: warning: converting to non-pointer type 'uint8_t {aka unsigned char}' from NULL [-Wconversion-null]
   return NULL;
          ^
.pio\libdeps\esp32dev\WiFi_ID870\src\utility\wifi_drv.cpp: In static member function 'static int32_t WiFiDrv::getRSSINetoworks(uint8_t)':
.pio\libdeps\esp32dev\WiFi_ID870\src\utility\wifi_drv.cpp:476:10: warning: converting to non-pointer type 'int' from NULL [-Wconversion-null]
   return NULL;
          ^
Archiving .pio\build\esp32dev\libc61\libSimpleBLE.a
.pio\libdeps\esp32dev\WiFi_ID870\src\utility\spi_drv.cpp: In static member function 'static char SpiDrv::spiTransfer(char)':
.pio\libdeps\esp32dev\WiFi_ID870\src\utility\spi_drv.cpp:36:78: warning: integer overflow in expression [-Woverflow]
 #define DELAY_SPI(X) { int ii=0; do { asm volatile("nop"); } while (++ii < (X*F_CPU/16000000)); }
                                                                              ^
.pio\libdeps\esp32dev\WiFi_ID870\src\utility\spi_drv.cpp:37:26: note: in expansion of macro 'DELAY_SPI'
 #define DELAY_TRANSFER() DELAY_SPI(10)
                          ^
.pio\libdeps\esp32dev\WiFi_ID870\src\utility\spi_drv.cpp:76:5: note: in expansion of macro 'DELAY_TRANSFER'
     DELAY_TRANSFER();
     ^
.pio\libdeps\esp32dev\WiFi_ID870\src\utility\spi_drv.cpp: In static member function 'static void SpiDrv::getParam(uint8_t*)':
.pio\libdeps\esp32dev\WiFi_ID870\src\utility\spi_drv.cpp:36:78: warning: integer overflow in expression [-Woverflow]
 #define DELAY_SPI(X) { int ii=0; do { asm volatile("nop"); } while (++ii < (X*F_CPU/16000000)); }
                                                                              ^
.pio\libdeps\esp32dev\WiFi_ID870\src\utility\spi_drv.cpp:37:26: note: in expansion of macro 'DELAY_SPI'
 #define DELAY_TRANSFER() DELAY_SPI(10)
                          ^
.pio\libdeps\esp32dev\WiFi_ID870\src\utility\spi_drv.cpp:148:5: note: in expansion of macro 'DELAY_TRANSFER'
     DELAY_TRANSFER();
     ^
*** [.pio\build\esp32dev\src\main.cpp.o] Error 1
In file included from .pio\libdeps\esp32dev\WiFi_ID870\src\utility\spi_drv.cpp:22:0:
.pio/libdeps/esp32dev/WiFi_ID870/src/utility/spi_drv.h: At global scope:
.pio/libdeps/esp32dev/WiFi_ID870/src/utility/spi_drv.h:41:13: warning: 'initialized' defined but not used [-Wunused-variable]
 static bool initialized = false;
             ^
C:\Users\Tony\.platformio\packages\framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6\libraries\ESPmDNS\src\ESPmDNS.cpp: In member function 'bool MDNSResponder::begin(const char*)':
C:\Users\Tony\.platformio\packages\framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6\libraries\ESPmDNS\src\ESPmDNS.cpp:68:10: error: 'class WiFiClass' has no member named 'onEvent'
     WiFi.onEvent(_on_sys_event);
```

> Library names are case-sensitive, so `Wifi` should be `WiFi`
Fixed Typo, still same.

valeros, 2020-06-25:
You need to delete previously downloaded library ".pio\libdeps\esp32dev\WiFi_ID870"

Tony763, 2020-06-25:
It's not there:
[image]

I had to delete `libdeps` and `build` directories. Also I had to move `WiFi` upward to second place and move `void scan_content_length` upward in code.

Now it runs OK. 
Last question, can I get ID or path from where they were installed?

valeros, 2020-06-25:
> Last question, can I get ID or path from where they were installed?

What do you mean? You can compile your project in verbose mode then the full paths to libraries will be printed in the dependency graph.

Tony763, 2020-06-25:
When installing new library from register I can see where it was taken.
```
Installing dependencies
Looking for Adafruit GFX Library library in registry
Found: https://platformio.org/lib/show/13/Adafruit GFX Library
LibraryManager: Installing id=13
Using cache: C:\Users\Tony\.platformio\.cache\cb\eb4a83205a6bf7664c071bda082bc1cb
Adafruit GFX Library @ 1.9.0 has been successfully installed!
```
So I can set them like this:
```
    13                      ;Adafruit GFX Library https://platformio.org/lib/show/13/Adafruit GFX Library
    571                     ;Adafruit ILI9341 https://platformio.org/lib/show/571/Adafruit ILI9341
    377                     ;Adafruit STMPE610 https://platformio.org/lib/show/377/Adafruit STMPE610
```
That's OK.

For libraries from framework I can find their names in `https://github.com/espressif/arduino-esp32.git`. But here is no ID so I have to set them by Name. For `SimpleBLE; WiFi; ESPmDNS; Wire; SPI; FS; Update;  ArduinoOTA` its ok but not for `SD` as it is not taken from framework, but from registry.