platformio/platform-espressif32 #370: DOIT ESP32 DEVKIT V1 doesn't mention Boot button

thenickdude, 2020-06-25:
Hey there,

This is the first Platform IO board I've ever used. I found that I was unable to upload a program to my DOIT ESP32 (got error message "OSError: [Errno 6] Device not configured") unless I was holding down the BOOT button on the board, or at least pressed it while it was hanging waiting to connect. Maybe this is obvious to experienced users, but could that detail be added to the info page about the board?

https://github.com/platformio/platformio-docs/blob/develop/boards/espressif32/esp32doit-devkit-v1.rst

valeros, 2020-06-25:
Hi @thenickdude ! Usually, we do not provide such details in our docs. But I'm interested whether the cause of this issue is hidden somewhere in our platform. Could you please try to upload firmware using Arduino IDE? Does it work there without holding the BOOT button.

thenickdude, 2020-06-25:
It behaves the same with the Arduino IDE, it hangs forever in the Connecting state until the BOOT button is held.

valeros, 2020-09-02:
Unfortunately, PlatformIO is not responsible for any issues relating to hardware.

closed 2020-09-02 by valeros