platformio/platform-espressif32 #373: SRC_FILTER option not working in build script scripts when running test

ChrisIdema, 2020-07-07:
I have a multi platform library that uses a python script to select sources and add build flags depended on the target.
I can use it as an external library in my project and it builds for both stm32 and esp32 (and in the future nrf).
However. When I run test on the library itself (with test_build_project_src=yes) the build fails for the esp32 target.
When runing the test command:
`platformio test -v -e esp32dev-4pin-test`
I get the following warning:
`Warning: the 'src_filter' option cannot be used with ESP-IDF. Select source files to build in the project CMakeLists.txt file.`
And I get errors because the compiler creates a default CMakeLists.txt-file in my src folder which recursively adds all source files (including src/stm32). It cannot build stm32 source files for the esp32 obviously.

My current workaround is to generate the CMakeLists.txt file in my src folder with my python script. Another workaround would be to commit a CMakeList.txt file to my repository. But I'd like to keep my build script in one place using the same python commands, without using CMake.

nathanjel, 2020-07-07:
@ChrisIdema I can suggest one workaround - don't use the "src" folder, but rename it to "srcfiles" or something else. Then CMake will not interfere and src_filter (actually `env.Append(SRC_FILTER=[...])` in library script to be specific) works for me.

ChrisIdema, 2020-07-09:
I was able to get it to work.

SRC_FILTER is relative to PROJECT_SRC_DIR . You can change PROJECT_SRC_DIR in test mode, **but it doesn't work if you use it as a library!**

In other words. This code **only** works in test mode:
```
src_filter = []
env.Replace(PROJECT_SRC_DIR=env['PROJECT_DIR'])  # set PROJECT_SRC_DIR to root, to avoid using "../"
src_filter.append("+<src/source_file_in_src_folder")
src_filter.append("+<other/source_file_in_other_folder")
```
But this code works in test mode **and** in library mode:
```
src_filter = []
src_filter.append("+<source_file_in_src_folder")
src_filter.append("+<../other/source_file_in_other_folder")
```
 
Include paths in build flags are **always** from root:
`env.ProcessFlags("-Iother/inc")`

closed 2020-07-09 by ChrisIdema