platformio/platform-espressif32 #377: Please add board TTGO-T-DISPLAY

DavidThijs, 2020-07-10:
What kind of issue is this?

- [X ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

### Configuration

**PlatformIO Version** (`platformio -- 4.3.4`):

### Hardware
Microcontroller ESPRESSIF-ESP32 240MHz Xtensa® single-/dual-core 32-bit LX6 microprocessor
Frequency | 240MHz
Flash | 4MB (32Mb)
RAM | 520KB
Display | IPS ST7789V 1.14 Inch
Vendor | TTGO
Upload Speed | 921600
CPU Frequency | 240Mhz (WiFi/BT)
Flash Frequency | 80Mhz
Flash Mode | QIO
Partition Scheme | Default 4MB with spiffs (1.2MB APP/1.5 SPIFFS)
Core Debug Level | None
PSRAM | Disabled

### Description of problem

Request to add the board to the list like the other TTGO boards.

### Additional info

This board needs a library for controlling the onboard display
https://github.com/Bodmer/TFT_eSPI
It also needs a specific configuration
https://github.com/Xinyuan-LilyGO/TTGO-T-Display

valeros, 2020-07-13:
Hi @DavidThijs ! Is this supported in Arduino framework?

XneddyX, 2020-07-13:
hi @valeros . yes see here....
[#209 ](https://github.com/platformio/platform-espressif32/issues/209)
pls add this board...

valeros, 2020-07-13:
Duplicate #209

closed 2020-07-13 by valeros