platformio/platform-espressif32 #383: Problems with SPIFFS on 1.12.4

GiampaMan, 2020-07-22:
Hi!

Maybe this issue is old, but I have the exact same issue (#69 ) but this is happening when I pass in PlatformIO from version 1.9.0 to 1.12.4 on ESP32 platform. The only way I realized that this was happening was when I saw errors when parsing on ArduinoJson. Then I just tried to print directly the SPIFFS file on serial port but it was not possible. I receive on serial port lot of questions signs. Example: "???????????????????????????????????????????????????????????". Can anyone knows how to do in this case?

I will appreciate any help, I'm a beginner.

Thanks

Coobik, 2020-07-23:
same here. cannot read files from spiffs, that were uploaded using platformio - upload file system image.
tried everything, like settings CONFIG_SPIFFS_META_LENGTH to 0 and 4, turning off CONFIG_SPIFFS_USE_MTIME etc. etc.
file exists on spiffs, i can open it, but ftell reports size of 0 bytes, and both read and fread return 0 bytes.

Pilooz, 2020-08-04:
Hello it seems I have the same Issue. 
Files exist on Spiffs : I can list them, I can open them in read mode, but I can’t read the data. 
@GiampaMan  what is your code to read files ?

closed 2022-09-20 by stale[bot]