platformio/platform-espressif32 #385: Undefined reference to `_binary_src_certs_pem_start'

walidamriou, 2020-07-27:
I want to use SSL, so I have put " certs.pem " in /src folder. In the main.c, I have put " `extern const char cert_start[] asm("_binary_src_certs_pem_start");` ". and in platformio.ini, I have put " `board_build.embed_txtfiles = src/certs.pem` ".
I have tried to build the code and clean before the build and the same error: 
...
```
Linking .pio/build/esp32dev/firmware.elf
/home/walidamriou/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/src/update.c.o:(.literal.update+0x4): undefined reference to `_binary_src_certs_pem_start'
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32dev/firmware.elf] Error 1
```
... 
**Any solutions or ideas to fix this problem?**

valeros, 2020-07-27:
Hi @walidamriou ! Could you try `_binary_certs_pem_start` instead of `_binary_src_certs_pem_start`.

walidamriou, 2020-07-27:
Hi @valeros , I have tried it but the same problem, see:

[
[image]
](url)

I noted that when I used the certificate from main.c we are putting the full certificate not just the beginning.

valeros, 2020-07-27:
Could you please clean your project, compile it in verbose mode and upload the log somewhere as a txt file?

walidamriou, 2020-07-27:
@valeros I cleaned and built it in verbose mode, I have got:

```
...
u/.platformio/packages/framework-espidf/components/lwip/include/apps/sntp -I/home/walidamriou/.platformio/packages/framework-espidf/components/lwip/lwip/src/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/lwip/port/esp32/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/lwip/port/esp32/include/arch -I/home/walidamriou/.platformio/packages/framework-espidf/components/vfs/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/esp_wifi/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/esp_wifi/esp32/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/esp_eth/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/efuse/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/efuse/esp32/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/app_trace/include /home/walidamriou/.platformio/packages/framework-espidf/components/mbedtls/mbedtls/library/x509.c
xtensa-esp32-elf-gcc -o .pio/build/esp32dev/esp-idf/mbedtls/mbedtls/library/x509_create.c.o -c -std=gnu99 -mlongcalls -Wno-frame-address -Wall -Wextra -W -Wdeclaration-after-statement -Wwrite-strings -Wlogical-op -Wshadow -Wmissing-declarations -Wmissing-prototypes -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -Wno-old-style-declaration -DPLATFORMIO=40304 -DARDUINO_ESP32_DEV -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -D_GNU_SOURCE -DIDF_VER=\"HEAD-HASH-NOTFOUND\" -DGCC_NOT_5_2_0 -DESP_PLATFORM -I.pio/build/esp32dev/config -I/home/walidamriou/.platformio/packages/framework-espidf/components/mbedtls/port/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/mbedtls/mbedtls/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/newlib/platform_include -I/home/walidamriou/.platformio/packages/framework-espidf/components/freertos/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/heap/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/log/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/soc/esp32/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/soc/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/esp_rom/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/esp_common/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/xtensa/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/xtensa/esp32/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/esp32/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/driver/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/esp_ringbuf/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/esp_event/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/tcpip_adapter/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/lwip/include/apps -I/home/walidamriou/.platformio/packages/framework-espidf/components/lwip/include/apps/sntp -I/home/walidamriou/.platformio/packages/framework-espidf/components/lwip/lwip/src/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/lwip/port/esp32/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/lwip/port/esp32/include/arch -I/home/walidamriou/.platformio/packages/framework-espidf/components/vfs/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/esp_wifi/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/esp_wifi/esp32/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/esp_eth/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/efuse/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/efuse/esp32/include -I/home/walidamriou/.platformio/packages/framework-espidf/components/app_trace/include /home/walidamriou/.platformio/packages/framework-espidf/components/mbedtls/mbedtls/library/x509_create.c
xtensa-esp32-elf-ar rc .pio/build/esp32dev/esp-idf/mbedtls/mbedtls/library/libmbedtls.a .pio/build/esp32dev/esp-idf/mbedtls/mbedtls/library/debug.c.o .pio/build/esp32dev/esp-idf/mbedtls/mbedtls/library/ssl_cache.c.o .pio/build/esp32dev/esp-idf/mbedtls/mbedtls/library/ssl_ciphersuites.c.o .pio/build/esp32dev/esp-idf/mbedtls/mbedtls/library/ssl_cli.c.o .pio/build/esp32dev/esp-idf/mbedtls/mbedtls/library/ssl_cookie.c.o .pio/build/esp32dev/esp-idf/mbedtls/mbedtls/library/ssl_srv.c.o .pio/build/esp32dev/esp-idf/mbedtls/mbedtls/library/ssl_ticket.c.o .pio/build/esp32dev/esp-idf/mbedtls/mbedtls/library/ssl_tls.c.o .pio/build/esp32dev/esp-idf/mbedtls/port/mbedtls_debug.c.o .pio/build/esp32dev/esp-idf/mbedtls/port/net_sockets.c.o
xtensa-esp32-elf-gcc -o .pio/build/esp32dev/esp-idf/mbedtls/mbedtls/library/x509_crl.c.o -c -std=gnu99 -mlongcalls -Wno-frame-address -Wall -Wextra -W -Wdeclaration-after-statement -Wwrite-strings -Wlogical-op -Wshadow -Wmissing-declarations -Wmissing-p [... cut ...] /mbedtls/library/libmbedcrypto.a .pio/build/esp32dev/esp-idf/mbedtls/mbedtls/library/libmbedtls.a .pio/build/esp32dev/esp-idf/mbedtls/mbedtls/library/libmbedx509.a -lcoexist -lcore -lespnow -lmesh -lnet80211 -lphy -lpp -lrtc -lsmartconfig -lhal -lstdc++ -lgcov -lc -lm -lgcc -Wl,--end-group
============================================== [FAILED] Took 947.18 seconds ==============================================
The terminal process "platformio 'run', '--verbose'" terminated with exit code: 1.
```

valeros, 2020-09-02:
Hi guys, sorry for the late reply. I pushed some changes together with updates for ESP-IDF v4.1 that could possibly fix this issue.

It would be great if you could retest your projects with the upstream version:
```
[env:upstream_develop]
platform = https://github.com/platformio/platform-espressif32.git
framework = esp-idf
board = ...
```

Symbol names should be without additional paths, e.g. `_binary_coap_ca_pem_start` for `coap_ca.pem`

StavrosD, 2021-02-02:
https://github.com/platformio/platform-espressif32/issues/480#issuecomment-771187787

closed 2021-02-04 by walidamriou