platformio/platform-espressif32 #389: Add support for M5Atom

doctortee, 2020-08-03:
(I don't know if this should be posted here, or on the main repo...)

PlatformIO actually supports M5Stack and M5Stick, but doesn't yet support M5Atom. The configuration seems to be similar, but it has a dedicated library located here :
https://github.com/m5stack/M5Atom

Would it be possible to add a dedicated board configuration for them ? Many thanks !

TimDeve, 2020-08-10:
For people like me who ended up here looking for how to use platformio with the atom, looking at the [M5 docs](https://docs.m5stack.com/#/en/core/atom_matrix)  you can see that the Arduino IDE board is the same as the M5Stick.

This example config works:
```ino
[env:atom]
platform = espressif32
board = m5stick-c
framework = arduino
upload_speed = 1500000
monitor_speed = 115200

lib_deps =
  M5Atom@0.0.1
  FastLED@3.3.3
```

szymoni, 2020-09-23:
Hi, were you able to connect M5Atom to WiFi on this board configuration (alternatively board=pico32)?

TimDeve, 2020-09-23:
Yeah, no problem with the wifi, you can check out my [project](https://github.com/TimDeve/m5light/blob/edfc2ae2c61891ab9083ff2b23bac051dd9a0040/platformio.ini#L26-L33) for an example.

eliabieri, 2020-11-23:
any updates on this?

commit ddab9bbb (2021-02-24): Update Arduino core to v1.0.5

Resolve #298
Resolve #329
Resolve #344
Resolve #359
Resolve #378
Resolve #384
Resolve #389
Resolve #390
Resolve #414
Resolve #420
Resolve #436
Resolve #440
Resolve #493
closed 2021-02-24 by valeros