platformio/platform-espressif32 #391: JTAG Upload not Working with ESP-PROG and ESP32-DevKitC-32E

ronin8600, 2020-08-08:
I'm getting the following error when trying to upload firmware using the ESP-PROG and ESP32-DevKitC-32E

`** Programming Started **
embedded:startup.tcl:449: Error: ** Programming Failed **
in procedure 'program_esp32' 
in procedure 'program_esp' called at file "C:/Users/<username>/.platformio/packages/tool-openocd-esp32/share/openocd/scripts/target/esp32.cfg", line 64
in procedure 'program_error' called at file "C:/Users/<username>/.platformio/packages/tool-openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 106
at file "embedded:startup.tcl", line 449
*** [upload] Error 1`

Modifying “C:\Users<username>.platformio\platforms\espressif32\builder\main.py” to add reset to the program commands (line 373 and 379) seems to address this issue.

Should this work without the reset?

RavenLite, 2020-11-10:
I met this problem too.
Any suggestions hopefully?

RavenLite, 2020-11-12:
You can try the absolute path

closed 2022-09-20 by stale[bot]