platformio/platform-espressif32 #393: esp32_exception_decoder not decoding

sblantipodi, 2020-08-18:
as title. I have my 
monitor_filters = esp32_exception_decoder, colorize
in my platformio.ini using an esp32 from lolin.

exceptions are not decoded:
```
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4008a3aa  PS      : 0x00060c33  A0      : 0x8008c23e  A1      : 0x3ffb1d70  
A2      : 0x003ffb80  A3      : 0x0000abab  A4      : 0xb33fffff  A5      : 0x00000001  
A6      : 0x00060a20  A7      : 0x0000cdcd  A8      : 0x0000abab  A9      : 0x3ffb1e60  
A10     : 0x000005b4  A11     : 0x00000001  A12     : 0x00060220  A13     : 0x3ffc2be0  
A14     : 0x05fffffe  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x003ffb80  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x4008a3aa:0x3ffb1d70 0x4008c23b:0x3ffb1da0 0x4008c8af:0x3ffb1dc0 0x40084a48:0x3ffb1de0 0x40084a7d:0x3ffb1e00 0x40084f41:0x3ffb1e20 0x4000beaf:0x3ffb1e40 0x40139667:0x3ffb1e60 0x40139639:0x3ffb1e80 0x40158d43:0x3ffb1ea0 0x401414d6:0x3ffb1ee0 0x400d5e85:0x3ffb1f00 0x400d2cb6:0x3ffb1f20 0x400d4d72:0x3ffb1f40 0x400dc911:0x3ffb1fb0 0x4008939d:0x3ffb1fd0
```

is there a fix to this? thanks

sblantipodi, 2020-08-28:
@wassfila thanks for the kind reply, I know it but I would like to use PlatformIO for the purpose since it recently added the feature.

valeros, 2020-09-02:
Hi @Tasssadar, any chance you could take a look at this issue?

Tasssadar, 2020-09-02:
Hello, the trace you posted there looks like it should match. Can you post longer output from the serial monitor, ideally including the output immediately after start of the serial monitor?

SinglWolf, 2020-09-23:
I have the same problem.
`D:\WORK\ESP-IDF\ESP32-Radiola>platformio device monitor -f esp32_exception_decoder
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters

Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

--- Miniterm on COM5  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
et�R�n��╗ҰL°'��5SH��rs�'�Š     �]�O�%UM�)I��t��E3�i║U}FA*�e1�S�%=O�JC���K�gsZ� 0                                                                                                                                �
�I���x�YHh�k_��i0��)}d���0,�E�v���0���_�.��x�b�d����0,�}�r�'���C���:T��      ���k�V��C!��:��ff���0���:�C!��:��ff���4���:�Lj��0�W�l L&���m��H���d��Q0��0�ɕn�L&�C!��:��00���0�k�:���*�Y���╔0��8��*�I (30) boot:
 ESP-IDF 3.40100.200827 2nd stage bootloader
I (30) boot: compile time 17:59:15
I (30) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (78) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (85) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (93) boot:  3 ota_0            OTA app          00 10 00010000 001c0000
I (100) boot:  4 ota_1            OTA app          00 11 001d0000 001c0000
I (108) boot:  5 device           unknown          40 00 00390000 00001000
I (115) boot:  6 device1          unknown          42 00 00391000 00001000
I (123) boot:  7 stations         unknown          41 00 00392000 00010000
I (130) boot:  8 hardware         WiFi data        01 02 003a2000 00002000
I (138) boot: End of partition table
I (142) boot_comm: chip revision: 1, min. application chip revision: 0
I (150) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x43408 (275464) map
I (278) esp_image: segment 1: paddr=0x00053430 vaddr=0x3ffb0000 size=0x03f48 ( 16200) load
I (286) esp_image: segment 2: paddr=0x00057380 vaddr=0x40080000 size=0x00404 (  1028) load
I (286) esp_image: segment 3: paddr=0x0005778c vaddr=0x40080404 size=0x0888c ( 34956) load
I (310) esp_image: segment 4: paddr=0x00060020 vaddr=0x400d0020 size=0xb04a8 (722088) map
I (623) esp_image: segment 5: paddr=0x001104d0 vaddr=0x40088c90 size=0x16594 ( 91540) load
I (685) boot: Loaded app from partition at offset 0x10000
I (685) boot: Disabling RNG early entropy source...
I (686) psram: This chip is ESP32-D0WD
I (692) spiram: Found 64MBit SPI RAM device
I (695) spiram: SPI RAM mode: flash 40m sram 40m
I (700) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (708) cpu_start: Pro cpu up.
I (711) cpu_start: Application information:
I (716) cpu_start: Project name:     ESP32-Radiola
I (722) cpu_start: App version:      2.0.1
I (726) cpu_start: Compile time:     Sep 23 2020 17:57:14
I (732) cpu_start: ELF file SHA256:  7ed2f88232160cb4...
I (738) cpu_start: ESP-IDF:          3.40100.200827
I (744) cpu_start: Starting app cpu, entry point is 0x400833a4
I (0) cpu_start: App cpu up.
I (1636) spiram: SPI SRAM memory test OK
I (1637) heap_init: Initializing. RAM available for dynamic allocation:
I (1637) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1643) heap_init: At 3FFBC6C0 len 00023940 (142 KiB): DRAM
I (1649) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1656) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1662) heap_init: At 4009F224 len 00000DDC (3 KiB): IRAM
I (1668) cpu_start: Pro cpu start user code
I (1673) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (1695) spi_flash: detected chip: generic
I (1695) spi_flash: flash io: dio
I (1695) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
[... 159 lines cut ...]
lib_dir = components
description = ESP32-Radiola

[env:esp-wrover-kit]
platform = https://github.com/platformio/platform-espressif32.git
board = esp-wrover-kit
framework = espidf
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
monitor_flags = 
	--echo
	--raw

board_build.partitions = partitions.csv
build_flags = 
	-D FILENAME=ESP32Radiola
	-DCONFIG_SPIRAM_CACHE_WORKAROUND
	-Iinclude
	-Icomponents/audio_player/src
	-Icomponents/tda7313
	-Icomponents/fifo/src
	-Icomponents/ucglib/src
	-Icomponents/xpt2046
extra_scripts = pre:extra_script.py`
Why isn't decoding working?

Gearme, 2020-10-19:
Same problem - added the `monitor_filters = esp32_exception_decoder` with `platform = https://github.com/platformio/platform-espressif32.git` but continue to get undecoded back traces.

Gearme, 2020-10-19:
My issue has been resolved - for reference: I was missing a `build_type = debug`.

ZweiEuro, 2021-02-21:
I am getting similar issues with the watchdog backtrace:

```
E (95132) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (95132) task_wdt:  - IDLE0 (CPU 0)
E (95132) task_wdt: Tasks currently running:
E (95132) task_wdt: CPU 0: btController
E (95132) task_wdt: CPU 1: IDLE1
E (95132) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x4013B303:0x3FFBE790 0x40082FF9:0x3FFBE7B0 0x40188A40:0x3FFB5E90 0x40188FF9:0x3FFB5EB0 0x4008FED5:0x3FFB5EE0 

E (95132) task_wdt: Print CPU 1 backtrace

Backtrace:0x40081DC3:0x3FFBED90 0x40082FF9:0x3FFBEDB0 0x4000BFED:0x3FFBC870 0x40090135:0x3FFBC880 0x4013B5A7:0x3FFBC8A0 0x4013B5B3:0x3FFBC8D0 0x4013B669:0x3FFBC8F0 0x40090B0D:0x3FFBC910 0x4008FED5:0x3FFBC930 

E (100132) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (100132) task_wdt:  - IDLE0 (CPU 0)
E (100132) task_wdt: Tasks currently running:
E (100132) task_wdt: CPU 0: btController
E (100132) task_wdt: CPU 1: IDLE1
E (100132) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x4013B303:0x3FFBE790 0x40082FF9:0x3FFBE7B0 0x40188A40:0x3FFB5E90 0x40188FF9:0x3FFB5EB0 0x4008FED5:0x3FFB5EE0 

E (100132) task_wdt: Print CPU 1 backtrace

Backtrace:0x40081DC3:0x3FFBED90 0x40082FF9:0x3FFBEDB0 0x4000BFED:0x3FFBC870 0x40090135:0x3FFBC880 0x4013B5A7:0x3FFBC8A0 0x4013B5B3:0x3FFBC8D0 0x4013B669:0x3FFBC8F0 0x40090B0D:0x3FFBC910 0x4008FED5:0x3FFBC930 

```

sblantipodi, 2021-04-20:
> build_type = debug

my issue has been solved with this too. closing it.

closed 2021-04-20 by sblantipodi