platformio/platform-espressif32 #399: Add board M5Stack Core 2

bendo, 2020-09-05:
Please add board [M5Stack Core 2](https://docs.m5stack.com/#/en/core/core2).

I try to use board **m5stack-core-esp32** which works quite ok, the problem is that Core 2 board has bigger Flash and Memory,
so when I try to upload bigger program to the board I got exception:
`Error: The program size (1345350 bytes) is greater than maximum allowed (1310720 bytes)`

So I switched to **m5stack-fire** board, which has same Flash and Memory, but there are then another issues related to i2c.

I also tried to create my own board based on Core, but then I've got this exception:
`/home/abc/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:185:26: fatal error: pins_arduino.h: No such file or directory` 

```
{
  "build": {
    "arduino":{
      "ldscript": "esp32_out.ld"
    },
    "core": "esp32",
    "extra_flags": "-DARDUINO_M5Stack_Core_ESP32",
    "f_cpu": "240000000L",
    "f_flash": "40000000L",
    "flash_mode": "dio",
    "mcu": "esp32",
    "variant": "m5stack_core_2"
  },
  "connectivity": [
    "wifi",
    "bluetooth",
    "ethernet",
    "can"
  ],
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "M5Stack Core 2",
  "upload": {
    "flash_size": "16MB",
    "maximum_ram_size": 6553600,
    "maximum_size": 16777216,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "http://www.m5stack.com",
  "vendor": "M5Stack"
}

```

Thank you

jesse-schein, 2020-09-13:
my board comes tomorrow so I will also be interested. I do know you can easily change the partition table which is documented here: [https://docs.platformio.org/en/latest/platforms/espressif32.html#partition-tables](https://docs.platformio.org/en/latest/platforms/espressif32.html#partition-tables)

We will probably have to make a custom one since there are not too many boards with 16Mb flash and 8MG ram.

bitoexsc, 2020-11-21:
@bendo I build success like this , I think it help

http://jokercatz.blogspot.com/2020/11/m5stack-core2-build-from-platformio.html

JokerCatz, 2020-11-21:
@bendo I build success like this , I think it help

http://jokercatz.blogspot.com/2020/11/m5stack-core2-build-from-platformio.html

bendo, 2020-12-13:
Thank you @JokerCatz I've tried it, and it works fine. I am now able to push bigger programs to my M5Core2 :+1:

closed 2020-12-13 by bendo