platformio/platform-espressif32 #400: Support for Espressif ESP32-DevKitC V4 board with ESP32-WROOM-32D module

kfine100, 2020-09-11:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ x] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

Support for Espressif ESP32-DevKitC V4. I have the version with the ESP32-WROOM-32D module. Not sure if any of the existing boards on PlatformIO corresond to this board. Thanks

valeros, 2020-09-11:
You should be able to use `esp32dev` board since the only difference is that `ESP32-WROOM-32D` has U.FL connector instead of an on-board antenna.

closed 2020-09-11 by valeros

cooley420, 2023-05-17:
esp32-wroom-32u has the antenna connector and 32d has on-board.