platformio/platform-espressif32 #401: esptool.py broken on macOS Big Sur Beta

PhilippMolitor, 2020-09-11:
This issue came with the PlatformIO 2.0 IDE update.

### Error Message
```
Traceback (most recent call last):
  File "/Users/phil/.platformio/packages/tool-esptoolpy/esptool.py", line 57, in <module>
    import serial.tools.list_ports as list_ports
  File "/Users/phil/.platformio/penv/lib/python3.8/site-packages/serial/tools/list_ports.py", line 29, in <module>
    from serial.tools.list_ports_posix import comports
  File "/Users/phil/.platformio/penv/lib/python3.8/site-packages/serial/tools/list_ports_posix.py", line 31, in <module>
    from serial.tools.list_ports_osx import comports
  File "/Users/phil/.platformio/penv/lib/python3.8/site-packages/serial/tools/list_ports_osx.py", line 32, in <module>
    kIOMasterPortDefault = ctypes.c_void_p.in_dll(iokit, "kIOMasterPortDefault")
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found
*** [.pio/build/debug/firmware.bin] Error 1
```

See: https://github.com/espressif/esptool/issues/540

hawtdawg, 2020-11-13:
> I can confirm this version working:
> https://github.com/espressif/esptool/blob/cb2060c0d768da38c9aa463e315e187a91f761f8/esptool.py
> 
> Maybe use this one for the current PlatformIO build instead?

Replaced current esptool.py with the ones you linked, but it's not working for me. It says:

`
Traceback (most recent call last):
  File "/Users/username/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 38, in <module>
    import serial
ImportError: No module named serial
exit status 1
Error compiling for board ESP32 Dev Module.
`

Any ideas?

gonzabrusco, 2020-11-15:
Check this out:

blackketter, 2020-11-18:
This is still an issue with the Big Sur official release and causes builds to fail as well as uploading.

valeros, 2020-11-19:
Hi guys, it would be great if someone could try to upload using the latest esptoolpy v3.0, for example:
```ini
[env:lolin32]
platform = espressif32
framework = arduino
board = lolin32
monitor_speed = 115200
platform_packages =
    platformio/tool-esptoolpy @ ~1.30000.0

```

blackketter, 2020-11-19:
That works great with my blink example.  Thanks @valeros

blackketter, 2020-11-19:
That is, it both builds and uploads correctly.

valeros, 2020-11-19:
Thanks!

esptoolpy v3.0 will be used by default in the next platform release. As for now users on Big Sur can override the package directly in their configuration:
```
platform_packages =
    platformio/tool-esptoolpy @ ~1.30000.0
```

closed 2020-11-19 by valeros

RobThree, 2020-11-20:
> esptoolpy v3.0 will be used by default in the next platform release

Any timeframe on that? Days? Weeks? Months?

valeros, 2020-11-20:
No later than the v1.0.5 release of Arduino core for ESP32.

RobThree, 2020-11-20:
> No later than the v1.0.5 release of Arduino core for ESP32.

I'm sorry, but I'm not familiar with any (if any?) release schedule for it. Is there one? (It's not the end of the world, at all, I'm just curious).

valeros, 2020-11-23:
There is not precise ETA. Usually, a new version of any platform is released after a new release of any supported frameworks.

mc-hamster, 2020-12-03:
I see this was included in the release notes of 2.1.0, published 9 hours ago.

commit 25e09897 (2020-11-19): Update esptoolpy to v3.0 // Resolve #401