platformio/platform-espressif32 #402: Possible to compile espidf-ulp-pulse example with framework = arduino, espidf ?

dumarjo, 2020-09-15:
Hi,
I try to use the ULP with this combinaison of framework = arduino, espidf. At first is not working because cmake output this:
```
Generating ULP configuration
CMake Error at C:/Users/myuser/.platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeDetermineSystem.cmake:99 (message):
  Could not find toolchain file:
  C:\Users\myuser\.platformio\packages\framework-espidf@3.40001.200521\components\ulp\cmake\toolchain-esp32-ulp.cmake

```
My platformio.ini look like this

```
[env]
platform = espressif32
framework = arduino, espidf
build_flags =
    -D CONFIG_BLINK_GPIO=2
monitor_speed = 115200
platform_packages =
; use a special branch
    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.0

[env:esp32dev]
board = esp32dev
```

From my search, the toolchain-esp32-ulp.cmake is in the ESPIDF but for version 4.1 and up. From my understanding is this .ini file working with version 4.0 of the ESPIDF. Is it possible to use a more recent of the the ESPIDF ?

If I copy the toolchain-ulp.cmake -> toolchain-esp32-ulp.cmake the error is removed.

The error is now error: implicit declaration of function 'rtc_io_number_get'

Again, I think this is because the ESPIDF is not updated. 

regards,
Jonathan

valeros, 2020-09-17:
At the moment, Arduino framework doesn't support recent versions (> 4.0) of ESP-IDF.

closed 2020-09-17 by valeros

dumarjo, 2020-09-17:
Hi,
with espressif32 @ 1.12.4 I get configure error. Cannot test it much...

`
PLATFORM: Espressif 32 (1.12.4) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40001.200521 (4.0.1)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
 - toolchain-xtensa32 2.80200.200827 (8.2.0)
Reading CMake configuration...
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.21.0.windows.1") 
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/myuser/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/myuser/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/myuser/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/myuser/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/myuser/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: 5655ee4-dirty
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies C:/Users/myuser/.platformio/packages/framework-espidf@3.40001.200521 ...
-- Configuring incomplete, errors occurred!
See also "D:/temp/espidf-ulp-pulse/.pio/build/esp32dev/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any of the parent directories): .git
  File "D:\scoop\apps\python\current\lib\site.py", line 178

    file=sys.stderr)

        ^

SyntaxError: invalid syntax
`
i'm on Windows 10 pro. 

Regards,
Jonathan

oshyshko, 2023-08-03:
I got a similar problem with macos + VS Code.
Any ideas?

```
[cmake] -- Building ESP-IDF components for target esp32
[cmake] -- Checking Python dependencies...
[cmake]   File "~/.platformio/packages/framework-espidf/tools/idf_tools.py", line 428
[cmake]     info(f'Downloading {url}')
[cmake]                             ^
[cmake] SyntaxError: invalid syntax
[cmake] CMake Error at ~/.platformio/packages/framework-espidf/tools/cmake/build.cmake:342 (message):
[cmake]   Some Python dependencies must be installed.  Check above message for
[cmake]   details.
[cmake] Call Stack (most recent call first):
[cmake]   ~/.platformio/packages/framework-espidf/tools/cmake/build.cmake:480 (__build_check_python)
[cmake]   ~/.platformio/packages/framework-espidf/tools/cmake/project.cmake:447 (idf_build_process)
[cmake]   CMakeLists.txt:3 (project)

```