platformio/platform-espressif32 #404: Unable to build combined arduino and idf project.

maxntf, 2020-09-16:
My OS is `Windows 10`. In the Visual Studio Code, the installed the `platformIO IDE`, in it I installed the `Espressif 32` package. Created a new project and select `Espressif IoT Development Framework` as the framework. In the project I edit the `platformio.ini` file.
```
[env:esp32cam]
platform = espressif32
board = esp32cam
framework = espidf, arduino
board_build.partitions = partitions_singleapp.csv
upload_speed = 921600
monitor_speed = 115200
; platform_packages = framework-arduinoespressif32, framework-espidf
```
Then I compile the project. My environment downloads another package to `C: \ Users \ Admin \ .platformio \ packages` with the name `framework-espidf@3.40001.200521`.
But in the end, `#include <Arduino.h>` is still not available in my `main.c`.
```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "sdkconfig.h"
#include <Arduino.h>

void app_main() { 
    while(1){
        vTaskDelay(1000 / portTICK_PERIOD_MS);
        printf("Hello World!");
    }
}
```
What am I doing wrong?
P.S.
I also created file `sdkconfig.h` in directory `src`, copied the contents from file `.pio\build\esp32cam\config\sdkconfig.h` there and added two directives to the end
```
#define ENABLE_ARDUINO_DEPENDS 1
#define CONFIG_AUTOSTART_ARDUINO 1
```

dumarjo, 2020-09-16:
Hi,

Have you tried this example ? https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-arduino-blink

Regards

Jonathan

maxntf, 2020-09-16:
> Hi,
> 
> Have you tried this example ? https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-arduino-blink
> 
> Regards
> 
> Jonathan

Yes, the example works for me.
But as soon as I create my new project, I see errors:
```
C:\users\admin\.platformio\packages\framework-arduinoespressif32-src-c69fc9322995db5694e2741d307d9723\libraries\WiFiClientSecure\src\ssl_client.cpp:23:4: error: #error "Please configure IDF framework to include mbedTLS -> Enable pre-shared-key ciphersuites and activate at least one cipher"
 #  error "Please configure IDF framework to include mbedTLS -> Enable pre-shared-key ciphersuites and activate at least one cipher"

C:\users\admin\.platformio\packages\framework-arduinoespressif32-src-c69fc9322995db5694e2741d307d9723\libraries\WiFiClientSecure\src\ssl_client.cpp:151:15: error: 'mbedtls_ssl_conf_psk' was not declared in this scope
```

maxntf, 2020-09-17:
I found the answer to my question [Using arduino-esp32 as a component fails on ESP-IDF v4.1](https://github.com/espressif/arduino-esp32/issues/3457#issuecomment-647102523)

valeros, 2020-09-17:
@maxntf Is there [sdkconfig.defaults](https://github.com/platformio/platform-espressif32/blob/develop/examples/espidf-arduino-blink/sdkconfig.defaults) in your project? You can execute `pio run -t menuconfig` in the terminal window to edit your project settings.

maxntf, 2020-09-30:
I still have questions.
In the `platformio.ini` file, I indicated that two frameworks would be used and ordered the download of the `framework-arduinoespressif32 v4.0` package. 
```
framework = arduino, espidf
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.0 
```
But I cannot use the `SDK libraries` from `\ .platformio \ packages \ framework-arduinoespressif32 \ tools \ sdk \ include`. I noticed that this patch is not written in the project file `.vscode /c_cpp_properties.json`. Only a patch for `core` and `libraries` is written here.
How can I implement this?

blmhemu, 2021-05-23:
You can rebuild the intellisense indexes to reflect changes in .vscode/c_cpp_properties.json

emdezla, 2021-11-26:
> 
> Then I compile the project. My environment downloads another package to `C: \ Users \ Admin \ .platformio \ packages` with the name `framework-espidf@3.40001.200521`. But in the end, `#include <Arduino.h>` is still not available in my `main.c`.
> 

Hello everyone, I am having the same problem here. I tried to follow the example previously mentioned  https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-arduino-blink but the <Arduino.h> file cannot be found. The funny thing is that the <Arduino.h> file is found when I select "framework = arduino" instead of "framework = arduino, espidf" inside platform.ini. 

I would be very grateful for any help you could provide me :)

Tom-Rfid, 2022-01-27:
The time to config your sdkconfig file with run>menuconfig comment your framework
;framework = arduino, espidf
framework = espidf

closed 2022-08-11 by stale[bot]