platformio/platform-espressif32 #405: Using custom framework-epsidf in platform_packages broken with v2.0.0

rohansingh, 2020-09-17:
I'm just getting around to upgrade to PIO core v5, and the latest platform-espressif32. I've been using a ESP-IDF fork that has patches for a couple components. Here's how it's setup:

```
[env]
platform = espressif32
framework = arduino, espidf
platform_packages =
  ; use our ESP-IDF fork
  ; unfortunately versioning this doesn't seem to work, so
  ; while we have a git commit hash here, the commit actually
  ; has to be pushed to `master`
  framework-espidf @ https://github.com/tidbyt/esp-idf.git#9ba64588cbc178088fb6ac30d6e21296ad32006d
```

This worked fine prior to upgrading, but it now fails very early in the build. Full build output is below, but the important error is:

```
CMake Error at CMakeLists.txt:5 (include):
  include could not find load file:

    /Users/rohan/.platformio/packages/framework-espidf/tools/cmake/project.cmake
```

Removing the `framework-espidf` override from `platform_packages` fixes this error. But that breaks my firmware which relies on some of the patches in that ESP-IDF fork.

The error seems to be stemming from this bog-standard line in my `CMakeLists.txt`:

```
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
```

This could be some kind of configuration error on my part but figured I'd open an issue in case anybody else has seen this.

---

Here's the build output:

```
PLATFORM: Espressif 32 (2.0.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (ftdi) On-board (ftdi) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.200129 (1.0.4)
 - framework-espidf 1.1.0+sha.9ba64588c
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
 - toolchain-xtensa32 2.80200.200827 (8.2.0)
Reading CMake configuration...
-- The C compiler identification is AppleClang 11.0.3.11030032
-- The CXX compiler identification is AppleClang 11.0.3.11030032
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring incomplete, errors occurred!
See also "/Users/rohan/tidbyt/firmware/.pio/build/tidbyt-v7/CMakeFiles/CMakeOutput.log".

CMake Error at CMakeLists.txt:5 (include):
  include could not find load file:

    /Users/rohan/.platformio/packages/framework-espidf/tools/cmake/project.cmake
```

rohansingh, 2020-09-17:
Digging a bit deeper, it loos like the `framework-espidf` package dir is empty for some reason:

```
[~/.platformio/packages/framework-espidf]$ ls
README.md              cjson-utils-tests.json package.json           spec_tests.json        tests.json             version.txt
```

valeros, 2020-09-17:
Hi @rohansingh ! The latest `espressif32` platform uses ESP-IDF v4.1 and if I'm not mistaken, Arduino framework only supports ESP-IDF v4.0. As a workaround you can always use the previous version of the `espressif32` platform:
```
[env:esp32dev]
platform = espressif32 @ 1.12.4
framework = ...
board = ...
```

rohansingh, 2020-09-17:
Hey @valeros, thanks for the hint but I don't think that's the underlying issue. I tried `espressif32 @ 1.12.4` and it didn't work.

It looks like the latest `espressif32` actually [detects the `arduino, esp-idf` case][1] and uses the older ESP-IDF in that case. But really it shouldn't matter, because I'm specifying my own ESP-IDF in `platform_packages`.

But for some reason, my `~/.platformio/packages/framework-espidf` directory ends up containing some random files from something called `json-patch-test-suite` (?) instead of my actual ESP-IDF fork. And then the build breaks since there is no actual ESP-IDF on my system anymore.

Really weird, though I guess technically this is not an issue with `platform-espressif32`. I can close this issue and open one in platformio/platformio-core if you'd like.

[1]: https://github.com/platformio/platform-espressif32/blob/develop/platform.py#L44

rohansingh, 2020-09-17:
Ah yup, looks like this is platformio/platformio-core#3624. Sorry.

closed 2020-09-17 by rohansingh