platformio/platform-espressif32 #408: Espidf-arduino-blink doesn't compile or run (unless CONFIG_AUTOSTART_ARDUINO=1)

ntwallace, 2020-09-20:
The ESP-IDF with Arduino as a component [blink example](https://github.com/platformio/platform-espressif32/blob/develop/examples/espidf-arduino-blink/src/Blink.cpp) does not compile unless `CONFIG_AUTOSTART_ARDUINO` is defined and it can find setup() and loop(). It compiles and runs as expected using the Arduino structure.

If `CONFIG_AUTOSTART_ARDUINO is not defined`, building fails with the error `.platformio/packages/framework-espidf@3.40001.200521/components/esp32/cpu_start.c:539: undefined reference to 'app_main'` because the file is .cpp and there is no extern for C.

Adding ```extern "C" {
    void app_main();
}``` to the example allows it to build, but results in a stack overflow/abort() when the RTOS scheduler is started by the CPU. With verbose logging, this looks like: 

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:7084
load:0x40078000,len:14080
load:0x40080400,len:4304
entry 0x400806e8
I (72) boot: Chip Revision: 3
I (72) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot: ESP-IDF 3.40001.200521 2nd stage bootloader
I (40) boot: compile time 14:51:09
I (40) boot: Enabling RNG early entropy source...
I (45) boot: SPI Speed      : 40MHz
I (49) boot: SPI Mode       : DIO
I (53) boot: SPI Flash Size : 16MB
I (57) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00011000 00004000
I (76) boot:  1 otadata          OTA data         01 00 00015000 00002000
I (83) boot:  2 phy_init         RF data          01 01 00017000 00001000
I (91) boot:  3 factory          factory app      00 00 00020000 00200000
I (98) boot:  4 ota_0            OTA app          00 10 00220000 00200000
I (106) boot:  5 ota_1            OTA app          00 11 00420000 00200000
I (113) boot: End of partition table
E (117) boot: ota data partition invalid, falling back to factory
I (124) boot_comm: chip revision: 3, min. application chip revision: 0
I (131) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x074d8 ( 29912) map
I (151) esp_image: segment 1: paddr=0x00027500 vaddr=0x3ffb0000 size=0x021f4 (  8692) load
I (155) esp_image: segment 2: paddr=0x000296fc vaddr=0x40080000 size=0x00400 (  1024) load
I (159) esp_image: segment 3: paddr=0x00029b04 vaddr=0x40080400 size=0x0650c ( 25868) load
I (178) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x174b4 ( 95412) map
I (212) esp_image: segment 5: paddr=0x000474d4 vaddr=0x4008690c size=0x03dc0 ( 15808) load
I (226) boot: Loaded app from partition at offset 0x20000
I (226) boot: Disabling RNG early entropy source...
I (227) cpu_start: Pro cpu up.
I (230) cpu_start: Application information:
I (235) cpu_start: Project name:     espidf-arduino-wifiscan
I (241) cpu_start: App version:      1
I (246) cpu_start: Compile time:     Sep 20 2020 14:50:05
I (252) cpu_start: ELF file SHA256:  f10f78b6b411f0b7...
I (258) cpu_start: ESP-IDF:          3.40001.200521
I (263) cpu_start: Starting app cpu, entry point is 0x40081394
I (250) cpu_start: App cpu up.
I (274) heap_init: Initializing. RAM available for dynamic allocation:
I (281) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (287) heap_init: At 3FFB32F0 len 0002CD10 (179 KiB): DRAM
I (293) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (299) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (306) heap_init: At 4008A6CC len 00015934 (86 KiB): IRAM
I (312) cpu_start: Pro cpu start user code
I (331) spi_flash: detected chip: gd
I (331) spi_flash: flash io: dio
I (331) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
***ERROR*** A stack overflow in task blink_task has been detected.
abort() was called at PC 0x4008a3a0 on core 0

ELF file SHA256: f10f78b6b411f0b7

Backtrace: 0x4008a009:0x3ffb71d0 0x4008a389:0x3ffb71f0 0x4008a3a0:0x3ffb7210 0x40087924:0x3ffb7230 0x40086b00:0x3ffb7250 0x40086ab6:0xa5a5a5a5 |<-CORRUPTED
  #0  0x4008a009:0x3ffb71d0 in invoke_abort at /Users/nick/.platformio/packages/framework-espidf@3.40001.200521/components/esp32/panic.c:155
  #1  0x4008a389:0x3ffb71f0 in abort at /Users/nick/.platformio/packages/framework-espidf@3.40001.200521/components/esp32/panic.c:172
  #2  0x4008a3a0:0x3ffb7210 in vApplicationStackOverflowHook at /Users/nick/.platformio/packages/framework-espidf@3.40001.200521/components/esp32/panic.c:123
  #3  0x40087924:0x3ffb7230 in vTaskSwitchContext at /Users/nick/.platformio/packages/framework-espidf@3.40001.200521/components/freertos/tasks.c:2770
  #4  0x40086b00:0x3ffb7250 in _frxt_dispatch at /Users/nick/.platformio/packages/framework-espidf@3.40001.200521/components/freertos/portasm.S:406
  #5  0x40086ab6:0xa5a5a5a5 in _frxt_int_exit at /Users/nick/.platformio/packages/framework-espidf@3.40001.200521/components/freertos/portasm.S:206

Rebooting...```

valeros, 2020-09-21:
Hi @ntwallace ! `CONFIG_AUTOSTART_ARDUINO` is set in the `sdkconfig.defaults` file: https://github.com/platformio/platform-espressif32/blob/develop/examples/espidf-arduino-blink/sdkconfig.defaults#L3

closed 2020-09-21 by valeros

ntwallace, 2020-09-21:
Indeed. I purposely turned it off, because I don't want the overhead of the loop task taking 4096 byes of the stack and running continuously on Core 1. 

My point was that the example breaks when Arduino autostart is turned off, and it shouldn't. 

Adding the extern for C and increasing the stack size of the tasks fixes it.

sylque, 2024-08-07:
I suggest this issue be reopened, as `espidf-arduino-blink` still doesn't run. The first problem (extern "C") has been fixed, but not the second one (configMINIMAL_STACK_SIZE).