platformio/platform-espressif32 #413: Update openocd

maxgerhardt, 2020-10-01:
The currently included version `1.1000.20190709` of `tool-openocd-esp32` is reported to have some problems with 

* ESP32 PICO + Ubuntu: https://community.platformio.org/t/debugging-esp32-pico-v4-1-works-on-win-10-but-fails-on-ubuntu-18-04-with-error-esp32-failed-to-read-insn-4/16291
* ESP32-WROOM-UE chips: https://community.platformio.org/t/jtag-debugging-with-esp32-wroom-32ue/16236

Both of which were reportedly solved by updating openocd to the the latest release ([v0.10.0-esp32-20200709](https://github.com/espressif/openocd-esp32/releases)). So, PIO should update too.

maxgerhardt, 2020-10-21:
Again a case of an outdated tool-openocd causing problems and being reported to be resolved by using the latest version above: https://community.platformio.org/t/cannot-debug-esp32/16644/9 

@valeros can you update the package? (:

valeros, 2020-12-02:
Resolved in https://github.com/platformio/platform-espressif32/commit/f5ffb5896c691634998a3c02eded961840a0bf66

closed 2020-12-02 by valeros