platformio/platform-espressif32 #416: Upload filesystem image OTA uploads to the wrong partition

programminghoch10, 2020-10-08:
When uploading a spiffs image using the option "Upload filesystem image OTA" the "--spiffs" flag is not passed to esp ota tool.
This results in the ota tool trying to upload to the program partition.
In my case it failed due to wrong magic header though.

PlatformIO Version: 5.0.4
OS: Windows 10

`platformio.ini`:
OTA related settings:
```ini
upload_protocol = espota
upload_port = esp.local
upload_flags =
  --port=3232
```
and framework related settings:
```
platform = https://github.com/platformio/platform-espressif32.git
board = esp32dev
framework = arduino
platform_packages = framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git
```

`upload log`:
```
Uploading .pio\build\esp32\spiffs.bin
22:01:46 [DEBUG]: Options: {'esp_ip': 'esp.local', 'host_ip': '0.0.0.0', 'esp_port': 3232, 'host_port': 26978, 'auth': '', 'image': '.pio\\build\\esp32\\spiffs.bin', 'spiffs': False, 'debug': True, 'progress': True, 'timeout': 10}
22:01:46 [INFO]: Starting on 0.0.0.0:26978
22:01:46 [INFO]: Upload size: 196608
Sending invitation to esp.local 
22:01:46 [INFO]: Waiting for device...

Uploading: [======================                                      ] 36%
22:01:46 [ERROR]: Error Uploading
*** [uploadfsota] Error 1
===================
[FAILED] Took 10.36 seconds 
===================
The terminal process "C:\Users\xxx\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'uploadfsota', '--environment', 'esp32'" terminated with exit code: 1.
```
Notice in `options`, `spiffs` is set to `False`!

thewavelength, 2020-10-24:
I have the same issue and also the same fix works for me. This should be fixed as it was very confusing.

heikkoe, 2020-12-09:
+1 this is very confusing, why have two menus, why the obvious one for OTA does not work and why are neither present in command palette

valeros, 2021-02-03:
Thanks for reporting, should be fixed in the development branch, please retest with the following configuration:
```ini
[env:ota_test]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32dev
framework = arduino
upload_protocol = espota
```

closed 2021-02-03 by valeros

imsunburnt, 2023-03-01:
This problem still persists for Upload Filesystem Image OTA in this version of Visual Studio Code that is targeting ESP8266 d1_mini:

Version: 1.75.1 (user setup)
Commit: 441438abd1ac652551dbe4d408dfcec8a499b8bf
Date: 2023-02-08T21:32:34.589Z
Electron: 19.1.9
Chromium: 102.0.5005.194
Node.js: 16.14.2
V8: 10.2.154.23-electron.0
OS: Windows_NT x64 10.0.19045
Sandboxed: No

Output:

Executing task in folder ttp223-contact-switch: C:\Users\dave\.platformio\penv\Scripts\platformio.exe run --target uploadfsota --environment d1_mini
...
Uploading .pio\build\d1_mini\spiffs.bin
21:29:49 [DEBUG]: Options: {'esp_ip': 'ttp223-contact-switch-2.local', 'host_ip': '0.0.0.0', 'esp_port': 8266, 'host_port': 52751, 'auth': '', 'image': '.pio\\build\\d1_mini\\spiffs.bin', 'spiffs': False, 'debug': True, 'progress': True}

Note the setting:  'spiffs': False

My platformio.ini file:

[platformio]
hostname = ttp223-contact-switch

[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
monitor_speed = 115200
upload_speed = 921600
build_flags = '-D HOSTNAME="${platformio.hostname}"' '-D MDNSNAME="${platformio.hostname}"' '-D OTANAME="${platformio.hostname}"'
monitor_port = COM[3]
upload_port = ${platformio.hostname}.local
upload_protocol = espota

The command Upload Filesystem Image does work fine and uploads using OTA.

commit a147b3c8 (2021-02-03): Add missing argument to OTA tool when uploading SPIFFS image
Resolve #416