platformio/platform-espressif32 #423: menuconfig up/down arrows don't work

maxgerhardt, 2020-10-27:
Issue #302 is again occuring. 

I'm using the latest PIO dev core `version 5.0.2b3` and the very latest development platform with `https://github.com/platformio/platform-espressif32.git`

The "L" key seems to be mapped to the Enter key too, "J" moves the cursor down and "K" moves the cursor up. "G" jumps to the top. Note that my keyboard layout is a german one, which may influence it (?).

Users report that normal invocation of `idf.py menuconfig` makes the arrow keys work as usual.

The command is invoked from a PIO terminal.

```
> pio run
..
Running menuconfig...
[0/1] cmd.exe /C "cd /D C:\Users\Maxi\Documents\PlatformIO\Projects\esp_test\.pio\build\esp32 && python C...nv --env IDF_TARGET=esp32 --output config C:/Users/Maxi/Documents/PlatformIO/Projects/esp_test/sdkconfig"Loaded configuration 'C:/Users/Maxi/Documents/PlatformIO/Projects/esp_test/sdkconfig'
No changes to save (for 'C:/Users/Maxi/Documents/PlatformIO/Projects/esp_test/sdkconfig')
..
```

maxgerhardt, 2020-10-30:
[This comment](https://community.platformio.org/t/esp32-idf-menuconfig-platofrmio-dragging-their-feet/6268/26?u=maxgerhardt) also seems interesting:

>In VSCode terminal you can simulate the up/down arrow keys by using the - and + keys on the numeric keypad/numpad, or - and + (shift + =) on the main part of the keyboard.

So maybe it's a misconfiguration of the VSCode terminal?

davidjade, 2020-11-03:
The - & + keys do not work for me, either on the numpad or the standard ones (nor do the arrows). However, the J/K keys work so at least there is something for now. 

I've also noticed that if I activate pio in a normal terminal outside VSCode (via %userprofile%\.platformio\penv\Scripts\activate ) and run pio run -t menuconfig, then all the keys work. It seems that VSCode is just eating the arrow keys when PIO is used.

maxgerhardt, 2020-12-17:
[This comment](https://community.platformio.org/t/how-to-add-kconfig-for-library/17856/5?u=maxgerhardt) is further interesting:

>PS. Some times my menuconfig start like your - yellow/gray UI. But in this UI Arrows keys not work. How to resolve? I use pio run -t manuconfig - then menu config starts in classik Kernel Config UI (blue/gray) where Arrow keys are works.

Never noticed that there are two possible versions?

Spider84, 2020-12-17:
[image]
[image]

ppotter100, 2021-01-25:
I'm having the same problem. Just freshly  installed PlatformIO on a new pc out of the box. Then get the yellow screen as well and the arrow keys don't work.

valeros, 2021-01-25:
Hi @ppotter100 ! Do `J` and `K` buttons work?

ppotter100, 2021-01-25:
Yes, they do work (I just read the documentation that stated the same somewhere.. -.-).

cooperg2001, 2022-04-11:
Just encountered this issue (after wondering why I couldn't use my arrow keys):
Keyboard Layout: QWERTY / US
Software Layout / Language: US English
OS: Windows 10 Edu

Can confirm that K and J have the same effect as the original issue comment states.

fervenceslau, 2022-06-15:
Thanks for the J/K tip, it really helped me. I don't know what kind of idiot would not program the arrow keys in this or at least leave a comment in the help screen to tell the user that J/K keys can be used to change the options.

Matteo2L, 2022-08-23:
Same issue, thank you very much for the tip!

drorgl, 2022-09-07:
The J/K tip worked but when searching in menus its not very effective which forced me to find the solution.

Unmark this checkbox and restart the terminals in vscode settings:

Terminal › Integrated: Windows Enable Conpty
[ ] Whether to use ConPTY for Windows terminal process communication (requires Windows 10 build number 18309+). Winpty will be used if this is false.

truedat101, 2022-09-29:
Not completely on the topic, worth mentioning on windows, definitely some difference in behavior depending on your shell, reliable is the old CMD prompt in a standalone window.  I got odd/incorrect behavior in any other shell (git bash, etc..).

peteDDD, 2023-10-18:
Let's expand on @drorgl comments...
"The J/K tip worked but when searching in menus its not very effective which forced me to find the solution."

When using the "/" command to find a menuconfig setting, such as "/wdt". The search will return a window listing all the config options that contain "wdt".  However, I can't find a way to move the cursor to that window to use j and k to scroll there and select one of those listed settings.   Pressing j or k just adds that letter onto the end of the search field, so, for example "/wdt" becomes "wdtj".   Using the mouse cursor to highlight a field in the list of search results does nothing to help enter that window.

I tried @drorgl approach of turning off ConPTY and restarting VSC and that did not help.

Anyone found a solution to this?
Thanks.

Lupum1001, 2023-11-26:
> I tried @drorgl approach of turning off ConPTY and restarting VSC and that did not help.
> 
> Anyone found a solution to this?

Disabling the setting and restarting VSCode as a whole wasn't enough; I had to hit the trashcan icon on the right to close the pio terminal itself, then re-open it. It did work after that though.