platformio/platform-espressif32 #425: Unable to compile espidf-aws-iot example

pkdhital, 2020-11-07:
compiling with recent platformio version (5.0.2) the example doesn't find embedded binary files.
Returns error:
.pio/build/esp32dev/esp-idf/src/subscribe_publish_sample.c.o:(.literal.aws_iot_task+0x20): undefined reference to `_binary_aws_root_ca_pem_start'

valeros, 2020-11-10:
Hi @pkdhital ! Please clean your project, compile it in verbose mode and attach the output here. Thanks!

StavrosD, 2021-02-02:
https://github.com/platformio/platform-espressif32/issues/480#issuecomment-771187787

cocus, 2021-02-16:
I've been trying to get the embedded files feature working using the aws-iot example as a basis for it, but it also doesn't work. I see that `board_build.embed_files` is being processed somewhat (I get a message saying `Generating assembly for .pio\build\esp32dev\favicon.ico.S`), however that file is either not being linked in, or the exported global symbol is not right. I had a look at that file, and I saw these contents:
```
/* * Data converted from src\favicon.ico
 * (null byte appended)
 */
.data
.section .rodata.embedded

.global favicon_ico
favicon_ico:

.global _binary_favicon_ico_start
_binary_favicon_ico_start: /* for objcopy compatibility */
/* [a bunch of .bytes], not included */

.global _binary_favicon_ico_end
_binary_favicon_ico_end: /* for objcopy compatibility */

.global favicon_ico_length
favicon_ico_length: /* not including null byte */
.word 6429
```
It looks fine to me; however I saw that it added a null byte at the end, which is wrong. No one asked for that (I'm not using `txtfiles`, but rather `files`); so that's another issue.

valeros, 2021-02-16:
@cocus What is your CMakeLists.txt in the `src` folder? Have you registered that files using `target_add_binary_data` function?

cocus, 2021-02-16:
I just checked the comment added by @StavrosD; that actually helped.
I was blindly following the docs where it's not stated I need to register the files using that function.
Would it be better if the configs set on platformio.ini do call that `target_add_binary_data`? Or at least specify that `board_build.embed_files` will "create an assembly .S file with the file contents, but it'll not add it to the link list".

closed 2022-08-10 by stale[bot]