platformio/platform-espressif32 #427: Question About base framework file versions

vito45h, 2020-11-09:
I have a project in VS code using platformio.ini, I have discovered some DHCP bugs, and I would like to report them to the correct repo, for now I'm just trying to understand where all the source codes are with my current configuration

My platformio.ini is set as such:
platform = espressif32@2.0.0
board = esp32dev
framework = arduino

I understand that in this setup the platform is coming from 
https://github.com/platformio/platform-espressif32/releases/tag/v2.0.0

with my framework set to arduino is it pulling the framework-arduinoespressif32 from
https://github.com/espressif/arduino-esp32

In the release notes for 2.0.0 am I missing something, does it indicate which release/tag arduino-esp32 is from, or is it always master Head?

vito45h, 2020-11-09:
I did come across this 
https://docs.platformio.org/en/latest/platforms/espressif32.html?utm_source=github&utm_medium=arduino-esp32#using-arduino-framework-with-staging-version

but if I'm using platform = espressif32@2.0.0
does it point to a specific tag/commit from https://github.com/espressif/arduino-esp32 
If so which one?

valeros, 2020-11-10:
Hi @vito45h !
We only support the stable version of the packages, so by default the `espressif32` platform uses Arduino ESP32 Core [v1.0.4](https://github.com/espressif/arduino-esp32/releases/tag/1.0.4)

You can try to specify the latest revision from the official repository, but there is no guarantee that it will work:
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
platform_packages =
    platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
```

closed 2020-11-10 by valeros

vito45h, 2020-11-16:
Would the following be equivalent to use in the platformio.ini?
platform = espressif32@2.0.0
or
platform_packages = platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#4638628873a061c36faffebe4d146d13f960076d
(I think 4638628873a061c36faffebe4d146d13f960076d is the commit for v1.0.4)

As a second question if I just use: platform = espressif32@2.0.0
Where does it actually download the .platformio/packages/framework-arduinoespressif32 files from?
I can't seem to figure out in this repo how platform.io knows where to get those files from.
Thanks

valeros, 2020-11-16:
> Would the following be equivalent to use in the platformio.ini?
platform = espressif32@2.0.0
or
platform_packages = platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#4638628873a061c36faffebe4d146d13f960076d
(I think 4638628873a061c36faffebe4d146d13f960076d is the commit for v1.0.4)

> platform_packages = platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#4638628873a061c36faffebe4d146d13f960076d

With this line you override the framework package the precise revision based on the commit. `platform = espressif32@2.0.0` is just the version of the platform that includes only stable packages.

> Where does it actually download the .platformio/packages/framework-arduinoespressif32 files from?
I can't seem to figure out in this repo how platform.io knows where to get those files from.

PlatformIO downloads packages from its own CDN.