platformio/platform-espressif32 #430: ESP-IDF compile error: pkg_resources cannot be imported

maxgerhardt, 2020-11-16:
When cloning this repository on a Ubuntu Budgie laptop and executing `pio run` in the `examples/espidf-aws-iot` folder, I'm presented with the following message.

``` 
$ pio run 
Processing esp32dev (platform: espressif32; framework: espidf; board: esp32dev)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2.0.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf 3.40100.200827 (4.1.0) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - tool-ninja 1.7.1 
 - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32 2.80200.200827 (8.2.0)
Reading CMake configuration...
-- Project version: v1.12.4-24-g07919d0
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
pkg_resources cannot be imported probably because the pip package is not installed and/or using a legacy Python interpreter. Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.
-- Configuring incomplete, errors occurred!
See also "/home/max/pio_tests/platform-espressif32/examples/espidf-aws-iot/.pio/build/esp32dev/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any of the parent directories): .git
CMake Error at /home/max/.platformio/packages/framework-espidf/tools/cmake/build.cmake:271 (message):
  Some Python dependencies must be installed.  Check above message for
  details.
Call Stack (most recent call first):
  /home/max/.platformio/packages/framework-espidf/tools/cmake/build.cmake:397 (__build_check_python)
  /home/max/.platformio/packages/framework-espidf/tools/cmake/project.cmake:396 (idf_build_process)
  CMakeLists.txt:6 (project)
```

Further going into file that throws this error message (`~/.platformio/packages/framework-espidf/tools/check_python_dependencies.py`) and adding a `print(sys.version)` statement reveals that in this file, the used Python interpreter is 

```
2.7.18 (default, Aug  4 2020, 11:16:42) 
[GCC 9.3.0]
```
I have installed PlatformIO globally using pip3 and everything else is working pretty fine. `python` maps to Python2, `python3` to Python3. 

```
$ pio system info
--------------------------  --------------------------------------------
PlatformIO Core             5.0.4a2
Python                      3.8.5-final.0
System Type                 linux_x86_64
Platform                    Linux-5.4.0-52-generic-x86_64-with-glibc2.29
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /home/max/.platformio
PlatformIO Core Executable  /home/max/.local/bin/platformio
Python Executable           /usr/bin/python3
Global Libraries            3
Development Platforms       6
Tools & Toolchains          58
--------------------------  --------------------------------------------
```
should I change something in the way I'm installing PlatformIO?

Matheus-Garbelini, 2020-11-27:
Hi. I am also getting the same issue. It was working well before, but now I'm getting `pkg_resources cannot be imported`

valeros, 2020-11-27:
Looks like the problem is caused by mixed Python installations in your system. The ESP-IDF build system is picking up the first available interpreter which might not be the right one. I pushed a small fix to force it use the interpreter where PlatformIO is installed. Would be great if someone with this issue could try to compile a blink project using the platform from the dev branch:
```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
framework = espidf
board = esp32dev
monitor_speed = 115200
```
Before compiling please uninstall the following packages:
```
pip3 uninstall pyparsing cryptography
```

maxgerhardt, 2020-11-27:
I uninstalled those pip libraries and modified that `platformio.ini` and got past the ESP-IDF error. 

```
$ pio run 
Processing esp32dev (platform: https://github.com/platformio/platform-espressif32.git; framework: espidf; board: esp32dev)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2.0.0+sha.cd0af53) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf 3.40100.200827 (4.1.0) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - tool-ninja 1.7.1 
 - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32 2.80200.200827 (8.2.0)
Reading CMake configuration...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Generating assembly for .pio/build/esp32dev/private.pem.key.S
Generating assembly for .pio/build/esp32dev/certificate.pem.crt.S
..
Linking .pio/build/esp32dev/firmware.elf
Retrieving maximum program size .pio/build/esp32dev/firmware.elf
Building .pio/build/esp32dev/firmware.bin
Checking size .pio/build/esp32dev/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  11.6% (used 37944 bytes from 327680 bytes)
Flash: [=======   ]  73.1% (used 766822 bytes from 1048576 bytes)
esptool.py v3.0
=========================================================================================== [SUCCESS] Took 86.21 seconds ===========================================================================================
```
Closing :)

closed 2020-11-27 by maxgerhardt

Matheus-Garbelini, 2020-11-28:
thanks a lot @valeros I've confirmed the fix. Is it possible to backport this to another version such as 1.12.4?
This is because my firmware relies on the SDK version that comes along with version 1.12.4 and I'm aware others may also have this issue but are stuck on an older SDK version.

maxgerhardt, 2020-11-28:
The commit that fixes it is a extremely simple 1 liner, see https://github.com/platformio/platform-espressif32/commit/cd0af53d4bb8773756107a12d8aee12c8b8c0bc8, so it should be trivial to either:

1. Manually modify  builder/frameworks/espidf.py of your installation, or 
2. Checkout this repository in commit a126f42f9118ea931c637fd7cb261f0bd474e20e (since this is the commit where 1.12.4 was released), apply the fix there, change the git remote to you repository if yours (or just copy the contents in your own github repository) and then point to it using `platform = <git url>`. 

The second way would be more portable. I'm not sure if @valeros can do a backport or modification of the 1.12.4 release.. at best he might be able to do a 1.12.5?

Matheus-Garbelini, 2020-12-01:
approach 1 works well if changing the global storage of platformio since it is predictable and does not require root.