platformio/platform-espressif32 #431: Increase looptask stack size

gonzabrusco, 2020-11-18:
Hello. I would like to increase the stack size of the FreeRTOS task that handles the Loop() funcion in Arduino. Can you please tell me the best way to achieve this in PlatformIO?

The file in question is this one: 

`https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/main.cpp
`
And I need to change this line:

`xTaskCreateUniversal(loopTask, "loopTask", 8192, NULL, 1, &loopTaskHandle, CONFIG_ARDUINO_RUNNING_CORE);
`
To this

`xTaskCreateUniversal(loopTask, "loopTask", 16384, NULL, 1, &loopTaskHandle, CONFIG_ARDUINO_RUNNING_CORE);
`
I can´t figure out the best way to do this. In Arduino IDE I could just modify the file in my local copy. But platformIO only shows me the object files (*.o) of the framework and no the source. I thought about cloning your repo, but you framework repository does no have any code (I guess the py scripts download directly from the official repository?).

I'm willing to work with PlatformIO but I'm lost at this point. How can I modify the framework to my liking? 

Thanks!

valeros, 2020-11-18:
Hi @gonzabrusco ! You can find your local copy here:
```
Unix ~/.platformio/packages/framework-arduinoespressif32
Windows %HOMEPATH%\.platformio\packages\framework-arduinoespressif32
```
Keep in mind that it might be overridden with the next platform release.

closed 2020-11-18 by valeros

gonzabrusco, 2020-11-18:
Can you please explain me why version 1.0.4 taken from espressif repository does not compile? 

I would like to make my own repository with my own changes. But for that first I need to understand why version 1.0.4 is not compiling. Thanks.

valeros, 2020-11-18:
> I would like to make my own repository with my own changes. But for that first I need to understand why version 1.0.4 is not compiling. Thanks.

What is the error message exactly?

gonzabrusco, 2020-11-18:
It's on my original post. I did not expressed myself well. I meant it does not finish downloading the platform_packages because it tries to download another repository (AzureIoT) and because the version does not match it aborts. But the version specifier was for the espressif repo, not for all of them. I think this is related to the .gitmodules file in the espressif repository.

valeros, 2020-11-19:
You need to use a tagged version of the repository:
```
platform_packages =
    platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#1.0.4
```
But keep in mind that the PlatformIO build script in that version is outdated, so you will need to use actual one from the original `framework-arduinoespressif32` package.

gonzabrusco, 2020-11-19:
Thank you @valeros !! Cloning the espressif repo, branching 1.0.4 and cherry picking your platformio-build.py commits did the trick. I can now make my own modifications to the stack. Thanks for you help.