platformio/platform-espressif32 #432: Esp32-S2 support

retsel89, 2020-11-18:
Hello, 
anyone knows dates for ESP32-S2 support on platform IO

GiovanniCmpaner, 2020-11-27:
If you are happy with Arduino framework and not esp-idf, here is an ini example:
```ini
[env:esp32doit-devkit-v1]
platform_packages = platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.2
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
board_build.mcu = esp32s2
```

retsel89, 2020-11-27:
Hi @GiovanniCmpaner, thanks for you answer.

That example from platform.ini throws the following message:

Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (2.0.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 0.0.0+sha.93c6226 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-esp32s2ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32s2 1.80200.200827 (8.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 29 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Linking .pio/build/esp32doit-devkit-v1/firmware.elf
xtensa-esp32s2-elf-g++: error: Drive/Platform_projects/testS2/.pio/build/esp32doit-devkit-v1/testS2.map: No such file or directory
*** [.pio/build/esp32doit-devkit-v1/firmware.elf] Error 1
========================================================================================= [FAILED] Took 1.62 seconds =========================================================================================
The terminal process "pio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

Missing the .map file, I think it is something related to the linker.
i compile an empty arduino setup - loop
how can I generate that .map file

GiovanniCmpaner, 2020-11-27:
Can you compile the simplest project, with empty setup and loop?
Does your project have spaces in the complete path?
Did you modify an existing `platformio.ini` using my example? What is the before and after file? Post here.

retsel89, 2020-11-27:
Wow. compile OK

i dont know why my default google drive folder have an space just in Drive word

/Volumes/Data/Google\ Drive/Platform_projects/testS2

i changed the project folder to a secure non space path and works. Thank you man. You save my life.

septatrix, 2020-12-10:
As mentioned in https://github.com/platformio/platform-espressif32/issues/349#issuecomment-742565874 a update of the bundled ESP-IDF version should add support.

fracool, 2021-01-14:
I'd be really interested to hear if anyone had any luck getting these boards working with platformio?

misery, 2021-01-14:
I can use my ESP32-S2 with esphome (platformio based) with this changes. Should be portable to your platformio usage, too.

https://github.com/esphome/esphome/pull/1433

valeros, 2021-01-25:
Hi guys, I've pushed updates for ESP-IDF v4.2, but at the moment without support for mixed Arduino, ESP-IDF projects. In case someone wants to try it out, please retest with the upstream version of the platform:
```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
framework = espidf
board = esp32dev
monitor_speed = 115200
```

closed 2021-01-25 by valeros