platformio/platform-espressif32 #433: Serial Monitor Resets ESP32 Dev Kit

gonzabrusco, 2020-11-19:
Hello. When I'm using the serial monitor to debug an ESP32 code, everything works as expected but when I close the terminal or finish it using Ctrl+C, it reboots de ESP32. I believe it is doing something with the RTS and DTS lines of the serial port that is resetting the ESP32 kit on serial disconnection. 

Can you tell me how to change this behaviour? I want to be able to disconnect the serial monitor without having a reboot. Now I'm forced to use Arduino IDE Serial Monitor because that one does not make the ESP32 reboot when closing it. 

Thanks
Gonzalo

valeros, 2020-11-19:
Hi @gonzabrusco !
I'm not sure how it's implemented on the Arduino IDE side, but you can take a look at the [monitor_rts](https://docs.platformio.org/en/latest/projectconf/section_env_monitor.html#monitor-rts) option. Also it might be useful to asses all available serial monitor options [here](https://docs.platformio.org/en/latest/core/userguide/device/cmd_monitor.html#cmd-device-monitor).

gonzabrusco, 2020-11-19:
Yes I was about to write here to close the issue. I've found out that forcing both RTS and DTR to zero stops resetting the ESP32. 

```
monitor_rts = 0
monitor_dtr = 0 
```
Thanks!

closed 2020-11-19 by gonzabrusco