platformio/platform-espressif32 #434: Add FTDI upload support for custom ESP32 Boards

devrim-oguz, 2020-11-20:
Hello, 

The PlatformIO IDE supports ESP development boards that use DTR and RTS lines for controlling the Enable and Boot pins. However, most of the FTDI adapters come with the lines DTR and CTS. If you make a custom board and try to program it with those cheap FTDI adapters, you have to manually reset the card. For this reason, would it be possible for you to add FTDI auto reset support for the PlatformIO IDE?

There are some existing tools to be used for this, for example this python code:
[](https://github.com/jimparis/esptool-ftdi)

Maybe you can add a project file option for using the FTDI auto reset mode, using this script in the IDE

Thanks in advance

valeros, 2020-11-20:
You can write your custom upload command using the [`upload_command`](https://docs.platformio.org/en/latest/projectconf/section_env_upload.html#upload-command) option.

devrim-oguz, 2020-11-20:
The script requires the esptool.py's location to be able to call it. How can I integrate it in the IDE? Thanks

valeros, 2020-11-20:
Just run upload command in verbose mode via the `pio run -t upload -v` command and you'll see the precise location of `esptool.py` script.

devrim-oguz, 2020-11-24:
Thanks for the response. I'm having a bit of difficulty creating the right upload command for it. Is there another way? Like creating an upload tool or something? Or can you help me construct an automatic upload command?

`AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa`

It says the upload command is this:

`"/home/ado/.platformio/penv/bin/python" "/home/ado/.platformio/packages/tool-esptoolpy/esptool.py" --chip esp32 --port "/dev/ttyUSB0" --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 /home/ado/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/bin/bootloader_qio_80m.bin 0x8000 /home/ado/Projects/PlatformIO/AutomatController/.pio/build/esp32doit-devkit-v1/partitions.bin 0xe000 /home/ado/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/partitions/boot_app0.bin 0x10000 .pio/build/esp32doit-devkit-v1/firmware.bin`

How can I make it fill the parts necessary automatically?

Espool-ftdi specifies that:
`usage: ./esptool-ftdi.py <path-to-esptool.py> [args...]`

Also I think it would be a very good addition to the project if there was an esptool-ftdi upload option. I think many people uses ftdi adapters to upload code. Thanks in advance.

valeros, 2020-11-24:
> Thanks for the response. I'm having a bit of difficulty creating the right upload command for it. Is there another way? Like creating an upload tool or something? Or can you help me construct an automatic upload command?

As an alternative, you can use an extra script, for example:
```ini
[env:lolin32]
platform = espressif32
framework = arduino
board = lolin32
monitor_speed = 115200
extra_scripts = esptool-ftdi.py

```
esptool-ftdi.py:
```python
Import("env")

# Move esptoolpy to upload flags
env.Prepend(UPLOADERFLAGS=[env.subst("$UPLOADER")])

# Use FTDI uploader
env.Replace(UPLOADER="/path/to/esptool-ftdi.py")
```

devrim-oguz, 2020-11-24:
Thank you! Will try.

valeros, 2021-01-21:
Feel free to reopen if you still need help.

closed 2021-01-21 by valeros