platformio/platform-espressif32 #438: Support for esp32s2

GiovanniCmpaner, 2020-11-27:
I'm trying to achieve esp32s2 (Arduino as component) support using the following `platformio.ini`:
```ini
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = espidf

monitor_speed = 115200
build_type = debug
board_build.speed = 921600
board_build.partitions = partitions_custom.csv
board_build.mcu = esp32s2
```
But I'm getting the following error:
<details>
<summary>
Failed to resolve component 'tinyusb' <b>(click for full log)</b>
</summary>

```
Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: espidf)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------Tool Manager: Installing platformio/framework-espidf @ ~3.40100.0
Unpacking  [####################################]  100%
Tool Manager: framework-espidf @ 3.40100.200827 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (2.0.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40100.200827 (4.1.0)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - tool-ninja 1.9.0
 - toolchain-esp32s2ulp 1.22851.191205 (2.28.51)
 - toolchain-xtensa32s2 1.80200.200827 (8.2.0)
Reading CMake configuration...
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.22.0.windows.1") 
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/Giovanni/.platformio/packages/toolchain-xtensa32s2/bin/xtensa-esp32s2-elf-gcc.exe
-- Check for working C compiler: C:/Users/Giovanni/.platformio/packages/toolchain-xtensa32s2/bin/xtensa-esp32s2-elf-gcc.exe
-- Check for working C compiler: C:/Users/Giovanni/.platformio/packages/toolchain-xtensa32s2/bin/xtensa-esp32s2-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/Giovanni/.platformio/packages/toolchain-xtensa32s2/bin/xtensa-esp32s2-elf-g++.exe
-- Check for working CXX compiler: C:/Users/Giovanni/.platformio/packages/toolchain-xtensa32s2/bin/xtensa-esp32s2-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: 6fb69a0-dirty
-- Building ESP-IDF components for target esp32s2beta
-- Checking Python dependencies...
Python requirements from C:\Users\Giovanni\.platformio\packages\framework-espidf\requirements.txt are satisfied.

-- Configuring incomplete, errors occurred!
See also "C:/Users/Giovanni/Documents/PlatformIO/Projects/auto2esp32/.pio/build/esp32doit-devkit-v1/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any of the parent directories): .git
CMake Error at C:/Users/Giovanni/.platformio/packages/framework-espidf/tools/cmake/build.cmake:186 (message):
  Failed to resolve component 'tinyusb'.
Call Stack (most recent call first):
  C:/Users/Giovanni/.platformio/packages/framework-espidf/tools/cmake/build.cmake:217 (__build_resolve_and_add_req)
  C:/Users/Giovanni/.platformio/packages/framework-espidf/tools/cmake/build.cmake:426 (__build_expand_requirements)
  C:/Users/Giovanni/.platformio/packages/framework-espidf/tools/cmake/project.cmake:396 (idf_build_process)
  CMakeLists.txt:3 (project)

===================================================================== [FAILED] Took 39.23 seconds =====================================================================
```
</details>

Apparently this `tinyusb` module is dependency from `esp32s2`, that exists on the latest esp-idf.
So I tried the latest esp-idf and PlataformIO branches using the following `platformio.ini`:
```ini
[env:esp32doit-devkit-v1]
platform_packages = framework-espidf @ https://github.com/espressif/esp-idf.git#release/v4.2
platform = https://github.com/platformio/platform-espressif32.git
board = esp32doit-devkit-v1
framework = espidf

monitor_speed = 115200
build_type = debug
board_build.speed = 921600
board_build.partitions = partitions_custom.csv
board_build.mcu = esp32s2
```
<details>
<summary> The C / CXX compiler identification is unknown <b>(click for full log)</b>
</summary>

```
Processing esp32doit-devkit-v1 (platform: https://github.com/platformio/platform-espressif32.git; board: esp32doit-devkit-v1; framework: espidf)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (2.0.0+sha.cd0af53) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 1.1.0+sha.511965b
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-ninja 1.9.0
 - toolchain-esp32s2ulp 1.22851.191205 (2.28.51)
 - toolchain-xtensa32s2 1.80200.200827 (8.2.0)
Reading CMake configuration...
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- Configuring incomplete, errors occurred!
See also "C:/Users/Giovanni/Documents/PlatformIO/Projects/auto2esp32/.pio/build/esp32doit-devkit-v1/CMakeFiles/CMakeOutput.log".
See also "C:/Users/Giovanni/Documents/PlatformIO/Projects/auto2esp32/.pio/build/esp32doit-devkit-v1/CMakeFiles/CMakeError.log".

CMake Error at CMakeLists.txt:2 (include):
  include could not find load file:

    C:\Users\Giovanni\.platformio\packages\framework-espidf/tools/cmake/project.cmake

CMake Error at CMakeLists.txt:3 (project):
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:3 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

===================================================================== [FAILED] Took 2.77 seconds =====================================================================
```
</details>

Apparently PlatformIO doesn't support the release/v4.2 from esp-idf.
What would be the solution to make it work with `esp32s2 (Arduino as component)`?

valeros, 2020-11-30:
We only support the stable versions of frameworks and ESP-IDF v4.2 hasn't been release yet.

closed 2020-11-30 by valeros

misery, 2020-12-26:
4.2 is released now. :-)
https://github.com/platformio/platform-espressif32/issues/455