platformio/platform-espressif32 #439: zephyr support

waspinator, 2020-11-27:
Is there a roadmap for zephyr support?

https://docs.zephyrproject.org/latest/boards/xtensa/esp32/doc/index.html

valeros, 2020-11-30:
Hi @waspinator ! As far as I know, the ESP32 port is still not production ready. Have you tried it with native Zephyr build tools? Does it work reliably?

H1ghBre4k3r, 2020-12-01:
I was able to flash the "Hello, World!"-example of Zephyr onto the ESP32 using `west`. 
Can test more projects and examples, so that we can test, if it works reliably.

silikone, 2021-01-21:
Waste of time unless you want to port the majority of the Zephyr features to ESP32 yourself. 
The current port is only very basic.

alvarowolfx, 2021-03-12:
Now Zephyr has WiFi station support on Zephyr. There are any plans to add bring Zephyr support with this ? 

Link to the PR: https://github.com/zephyrproject-rtos/zephyr/pull/32081/files

beriberikix, 2021-08-31:
Coming back to this - we have been using ESP32 now [reliably](https://youtu.be/H4MSrss3mSA) since Espressif has added official support. See more [here](https://github.com/zephyrproject-rtos/zephyr/issues/29394).

Szewcson, 2022-03-23:
Also interested in that feature. Espressif starts official support for Zephyr so we can expect that it will be a complete solution soon. It will be great if it will be also supported by platformio.

Anprotaku, 2024-12-04:
Zephyr is actually quite stable on esp32 devices now, any progress/base on this?

camiloVargasTechnaid, 2025-01-21:
Hello, zephyr is working good on ESP32, any plan to work on this with PlatformIO?