platformio/platform-espressif32 #441: Cannot build generated ESP32 project

pmarkiewicz, 2020-12-01:
Latest pio, created new ESP32  project from pio main page and one can't be build:

`Executing task: C:\Users\piotr_markiewicz\.platformio\penv\Scripts\pio.exe run --environment esp32doit-devkit-v1 <

Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: espidf)
---------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (2.0.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40100.200827 (4.1.0)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-ninja 1.9.0 
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
 - toolchain-xtensa32 2.80200.200827 (8.2.0)
Reading CMake configuration...
Error: Couldn't find target config target-__idf_cbor-0798204ef8a9c1caf973.json`

same error when I try to run task menuconfig

Schrolli91, 2020-12-05:
I have the exact same error.
Created a new ESP-IDF bases project and tried to build that empty project - no changes.

valeros, 2021-01-21:
That's strange, could you please try to clean your project and try again?

Schrolli91, 2021-01-21:
Have updated all (PIO, Toolchain, etc.) to their latest versions...
Now it builds ... No idea what happens last time

closed 2022-04-18 by valeros