platformio/platform-espressif32 #442: Unable to compile espidf-blinky project

maxgerhardt, 2020-12-03:
Reproduction:
* create a new directory (`mkdir espidf_test`) 
* copy the esp-idf blinky example to the folder (`cp -r C:\Users\Max\.platformio\platforms\espressif32\examples\espidf-blink\* .`)
* run the project (`pio run`) 

That gives the error message

```
> Executing task in folder espidf_testing: C:\Users\Max\AppData\Local\Programs\Python\Python38\Scripts\pio.exe run --environment esp32dev <

Processing esp32dev (platform: espressif32; board: esp32dev; framework: espidf)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2.0.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40100.200827 (4.1.0)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
 - toolchain-xtensa32 2.80200.200827 (8.2.0)
Reading CMake configuration...
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.29.2.windows.2") 
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/Max/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/Max/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/Max/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe -- broken
-- Configuring incomplete, errors occurred!
See also "C:/Users/Max/Documents/PlatformIO/Projects/espidf_testing/.pio/build/esp32dev/CMakeFiles/CMakeOutput.log".
See also "C:/Users/Max/Documents/PlatformIO/Projects/espidf_testing/.pio/build/esp32dev/CMakeFiles/CMakeError.log".

fatal: not a git repository (or any of the parent directories): .git
CMake Error at C:/Users/Max/.platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:44 (try_compile):
  Failed to open

    C:/Users/Max/Documents/PlatformIO/Projects/espidf_testing/.pio/build/esp32dev/CMakeFiles/CMakeTmp/CMakeLists.txt

  Permission denied
Call Stack (most recent call first):
  C:/Users/Max/.platformio/packages/framework-espidf/tools/cmake/project.cmake:308 (__project)
  CMakeLists.txt:3 (project)

CMake Error at C:/Users/Max/.platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "C:/Users/Max/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe"

  is not able to compile a simple test program.

  It fails with the following output:

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  C:/Users/Max/.platformio/packages/framework-espidf/tools/cmake/project.cmake:308 (__project)
  CMakeLists.txt:3 (project)
```

Observations: 

1. The file ` C:/Users/Max/Documents/PlatformIO/Projects/espidf_testing/.pio/build/esp32dev/CMakeFiles/CMakeTmp/CMakeLists.txt` does not exist (empty folder). 
2. The compiler executable `C:/Users/Max/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe` seems executable at least when doing `--version`. 
3. VSCode says in the output tab (translated from german) `[3.12.2020, 15:22:22] The configuration with compilerPath "C:/Users/Maxi/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe" could not be resolved. Instead, "C:\Program Files\mingw-w64\x86_64-8.1.0-posix-seh-rt_v6-rev0\mingw64\bin\gcc.exe" will be used`. So somehow it's falling back to my MinGW installation
4. Removing the folder `C:/Users/Max/.platformio/packages/toolchain-xtensa32/` (and the `.cache`) to make it redownload the compiler makes no difference

System info (latest dev core):

```
>pio system info
--------------------------  --------------------------------------------------------------------------
PlatformIO Core             5.0.4b1
Python                      3.8.6-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\Max\.platformio
PlatformIO Core Executable  C:\Users\Max\AppData\Local\Programs\Python\Python38\Scripts\platformio.exe
Python Executable           c:\users\max\appdata\local\programs\python\python38\python.exe
Global Libraries            0
Development Platforms       7
Tools & Toolchains          41
--------------------------  --------------------------------------------------------------------------
```

with latest Espressif32 platform, of course. 

I'm really stumbled by this issue o_o.

valeros, 2020-12-03:
Hi @maxgerhardt !  Please attach here two files if they exist:
```
See also "C:/Users/Max/Documents/espidf_test/.pio/build/esp32dev/CMakeFiles/CMakeOutput.log".
See also "C:/Users/Max/Documents/espidf_test/.pio/build/esp32dev/CMakeFiles/CMakeError.log".
```
You mentioned VSCode, but I see that `platformio-core` is installed outside of the virtual environment shipped with PlatformIO IDE. Do you use any custom settings in PlatformIO IDE?

maxgerhardt, 2020-12-03:
The files are: 

[CMakeError.log](https://github.com/platformio/platform-espressif32/files/5636980/CMakeError.log)
[CMakeOutput.log](https://github.com/platformio/platform-espressif32/files/5636981/CMakeOutput.log)

>You mentioned VSCode, but I see that platformio-core is installed outside of the virtual environment shipped with PlatformIO IDE. Do you use any custom settings in PlatformIO IDE?

PlatformIO was installed using the system's `pip`. The VSCode PlatformIO extension is set to *not* use the builtin core as the only deviation. 

The error message in VSCode appears when opening the project additionally in VSCode, but it was created from the CLI.

maxgerhardt, 2020-12-03:
However, using VSCode's built-in core in the development version results in the same error.

```
> Executing task in folder espidf_test: C:\Users\Max\.platformio\penv\Scripts\pio.exe run <

Processing esp32dev (platform: espressif32; framework: espidf; board: esp32dev)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2.1.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40100.200827 (4.1.0)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
 - toolchain-xtensa32 2.80200.200827 (8.2.0)
Reading CMake configuration...
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.29.2.windows.2") 
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/Max/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/Max/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/Max/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe -- broken   
-- Configuring incomplete, errors occurred!
See also "C:/Users/Max/Documents/espidf_test/.pio/build/esp32dev/CMakeFiles/CMakeOutput.log".
See also "C:/Users/Max/Documents/espidf_test/.pio/build/esp32dev/CMakeFiles/CMakeError.log".

fatal: not a git repository (or any of the parent directories): .git
CMake Error at C:/Users/Max/.platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:44 (try_compile):
  Failed to open

    C:/Users/Max/Documents/espidf_test/.pio/build/esp32dev/CMakeFiles/CMakeTmp/CMakeLists.txt

  Permission denied
Call Stack (most recent call first):
  C:/Users/Max/.platformio/packages/framework-espidf/tools/cmake/project.cmake:308 (__project)
  CMakeLists.txt:3 (project)

CMake Error at C:/Users/Max/.platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "C:/Users/Max/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe"

  is not able to compile a simple test program.

  It fails with the following output:

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  C:/Users/Max/.platformio/packages/framework-espidf/tools/cmake/project.cmake:308 (__project)
  CMakeLists.txt:3 (project)

=================================================================================================== [FAILED] Took 9.33 seconds ===================================================================================================
```

But this there is magically more information in the `CMakeError.log` file 

```
Determining if the C compiler works failed with the following output:

Determining if the C compiler works failed with the following output:

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: C:/Users/Max/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe 
Build flags: -mlongcalls;-Wno-frame-address
Id flags:  

The output was:
1
xtensa-esp32-elf-gcc.exe: error: CMakeCCompilerId.c: No such file or directory
xtensa-esp32-elf-gcc.exe: fatal error: no input files
compilation terminated.

Determining if the C compiler works failed with the following output:

Determining if the C compiler works failed with the following output:

Determining if the C compiler works failed with the following output:

```

valeros, 2020-12-03:
That's strange. I see `Permission denied` an other strange error messages. Do you have an antivirus in your system?

maxgerhardt, 2020-12-03:
Solved by disabling Avast Free Antivirus v20.9.2437. Incredible.. antivirus programs have become too intrusive. Not even a notification that the AV is doing something to the files in the folder appears. Works from the shell and VScode.

closed 2020-12-03 by maxgerhardt