platformio/platform-espressif32 #448: Add support for WT32-ETH01

nullos007, 2020-12-09:
Hi, it's possible to add this board : WT32-ETH01 ( with WT32-S1 MCU).

Thanks.

SzczepanLeon, 2020-12-11:
Here is LAN8720 configuration for that board:

```
//GPIO16 = set to HIGH
//GPIO18 = ETH MDIO
//GPIO23 = ETH MDC
#define ETH_TYPE      ETH_PHY_LAN8720
#define ETH_CLKMODE   ETH_CLOCK_GPIO0_IN
#define ETH_ADDR      1
```

efirshik, 2020-12-26:
where find work yaml config for this board?

nullos007, 2020-12-26:
> 
> 
> Here is LAN8720 configuration for that board:
> 
> ```
> //GPIO16 = set to HIGH
> //GPIO18 = ETH MDIO
> //GPIO23 = ETH MDC
> #define ETH_TYPE      ETH_PHY_LAN8720
> #define ETH_CLKMODE   ETH_CLOCK_GPIO0_IN
> #define ETH_ADDR      1
> ```

Thanks for the reply 👍

mondbaron, 2022-04-15:
solved by pull request #689

valeros, 2022-05-16:
Added in [v4.2.0](https://github.com/platformio/platform-espressif32/releases/tag/v4.2.0)

closed 2022-05-16 by valeros