platformio/platform-espressif32 #449: Issue building and reading CMake configuration

tennisparty, 2020-12-10:
Hello,

I am trying to upload code to an AIthinker board using the steps outlined here: [(https://github.com/espressif/esp32-camera)] 

I am running MacOS catalina and Platformio in VS code. I get the following error when trying to build the code:

`> Executing task in folder 201205-100432-esp32cam: pio run <

Processing esp32cam (platform: espressif32; board: esp32cam; framework: espidf)
-----------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam.html
PLATFORM: Espressif 32 (2.1.0) > AI Thinker ESP32-CAM
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf 3.40100.200827 (4.1.0) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - tool-ninja 1.9.0 
 - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32 2.80200.200827 (8.2.0)
Reading CMake configuration...
AssertionError: :
  File "/Users/level04/.platformio/penv/lib/python3.8/site-packages/platformio/builder/main.py", line 170:
    env.SConscript("$BUILD_SCRIPT")
  File "/Users/level04/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/level04/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/level04/.platformio/platforms/espressif32/builder/main.py", line 224:
    target_elf = env.BuildProgram()
  File "/Users/level04/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/Users/level04/.platformio/penv/lib/python3.8/site-packages/platformio/builder/tools/platformio.py", line 61:
    env.ProcessProgramDeps()
  File "/Users/level04/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/Users/level04/.platformio/penv/lib/python3.8/site-packages/platformio/builder/tools/platformio.py", line 127:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "/Users/level04/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/Users/level04/.platformio/penv/lib/python3.8/site-packages/platformio/builder/tools/platformio.py", line 342:
    SConscript(env.GetFrameworkScript(f), exports="env")
  File "/Users/level04/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py", line 661:
    return method(*args, **kw)
  File "/Users/level04/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/level04/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/level04/.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 1177:
    ota_partition_params = get_partition_info(
  File "/Users/level04/.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 838:
    assert os.path.isfile(pt_path)
`

My Platformio.ini file is as follows:
`; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32cam]
platform = espressif32
board = esp32cam
framework = espidf
monitor_speed = 115200
 build_flags =
   -I../lib/esp32-camera
[env]
 lib_deps =
  esp32-camera`

And my cmake.lists file is:
`cmake_minimum_required(VERSION 3.16.0)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(201205-100432-esp32cam)`

Can anyone help?
Many thanks,
Sam

tennisparty, 2020-12-10:
It turns out this was an issue with my python installation in Catalina. I managed to fix the issue using the last steps in this tutorial (changing global python to a version managed in pyenv) -https://opensource.com/article/19/5/python-3-default-mac

tennisparty, 2020-12-10:
I have tried chaing the python version in the environment but this has not helped

niilves, 2020-12-11:
Hi! 

I have the same issue on windows. 
Creating new project crashes in file \.platformio\platforms\espressif32\builder\frameworks\espidf.py on asserting if partition file exists. On line 1177 it tries to get the "PARTITIONS_TABLE_CSV" and asserts as the file is not found...

Seems like the code was added recently by @valeros
https://github.com/platformio/platform-espressif32/commit/115f98151ed2e8328ee392c62752bd4678b8b5e5 

I added a print(pt_path) before the assert. The assert is looking for file "huge_app.csv", which is not present in the project directory.

I am new to platformio and esp32 and do not know how to fix the problem.

Regards,
Nigul

valeros, 2020-12-15:
Looks like you're trying to use a partition table that doesn't exist in ESP-IDF. Try to specify a default one directly in your platformio.ini:
```ini
[env:esp32cam]
platform = espressif32
board = esp32cam
framework = espidf
monitor_speed = 115200
 build_flags =
   -I../lib/esp32-camera
[env]
 lib_deps =
  esp32-camera

board_build.partitions = partitions_singleapp.csv

```

valeros, 2021-01-26:
Resolved in https://github.com/platformio/platform-espressif32/commit/1d1a7ca53a99b3561f7bf1998100542874ffe2b0

closed 2021-01-26 by valeros