platformio/platform-espressif32 #450: Trouble to compile a upstream esp-idf v4.1

jonathandreyer, 2020-12-13:
Hello,

I am trying to use a modified version of esp-idf (v4.1 + backported patch) to investigate/resolve the issue espressif/esp-idf#6024, but I have some trouble to do so.
I have cloned the esp-idf repository and added the `package.json` file (can be seen [here](https://github.com/jonathandreyer/esp-idf/tree/release/v4.1)). Unfortunately, I wasn't able to build the project with this custom framework. I have some error which can be seen [here](https://github.com/jonathandreyer/esp32-sim7000/runs/1544958210?check_suite_focus=true). The error log in verbose mode is:
```
xtensa-esp32-elf-gcc -o .pio/build/esp32dev/esp-idf/mbedtls/mbedtls/library/x509_create.c.o -c -std=gnu99 -mlongcalls -Wno-frame-address -Wall -Wextra -W -Wdeclaration-after-statement -Wwrite-strings -Wlogical-op -Wshadow -Wmissing-declarations -Wmissing-prototypes -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -Wno-old-style-declaration -DPLATFORMIO=50003 -DARDUINO_ESP32_DEV -DCONFIG_EXAMPLE_MODEM_DEVICE_SIM800=1 -DCONFIG_EXAMPLE_COMPONENT_MODEM_APN=\"CMNET\" -DCONFIG_EXAMPLE_MODEM_PPP_AUTH_USERNAME=\"\" -DCONFIG_EXAMPLE_MODEM_PPP_AUTH_PASSWORD=\"\" -DCONFIG_EXAMPLE_MODEM_UART_TX_PIN=27 -DCONFIG_EXAMPLE_MODEM_UART_RX_PIN=26 -DCONFIG_EXAMPLE_MODEM_UART_RTS_PIN=0 -DCONFIG_EXAMPLE_MODEM_UART_CTS_PIN=0 -DCONFIG_EXAMPLE_MODEM_UART_EVENT_TASK_STACK_SIZE=2048 -DCONFIG_EXAMPLE_MODEM_UART_EVENT_TASK_PRIORITY=5 -DCONFIG_EXAMPLE_MODEM_UART_EVENT_QUEUE_SIZE=30 -DCONFIG_EXAMPLE_MODEM_UART_PATTERN_QUEUE_SIZE=20 -DCONFIG_EXAMPLE_MODEM_UART_TX_BUFFER_SIZE=512 -DCONFIG_EXAMPLE_MODEM_UART_RX_BUFFER_SIZE=1024 -DCONFIG_EXAMPLE_GPIO_MODEM_PWRKEY=4 -DCONFIG_EXAMPLE_GPIO_MODEM_RESET=5 -DCONFIG_EXAMPLE_GPIO_MODEM_STATUS=19 -DCORE_DEBUG_LEVEL=5 -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -D_GNU_SOURCE -DIDF_VER=\"3.40100.201112+sha.0dd94869a\" -DESP_PLATFORM -I.pio/build/esp32dev/config -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/mbedtls/port/include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/mbedtls/mbedtls/include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/newlib/platform_include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/freertos/include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/heap/include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/log/include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/lwip/include/apps -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/lwip/include/apps/sntp -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/lwip/lwip/src/include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/lwip/port/esp32/include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/lwip/port/esp32/include/arch -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/lwip/port/esp32/tcp_isn -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/soc/esp32/include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/soc/include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/esp_rom/include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/esp_common/include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/xtensa/include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/xtensa/esp32/include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/esp32/include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/driver/include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/driver/esp32/include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/esp_ringbuf/include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/efuse/include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/efuse/esp32/include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/vfs/include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/esp_wifi/include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/esp_wifi/esp32/include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/esp_event/include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/esp_netif/include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/esp_eth/include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/tcpip_adapter/include -I/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/app_trace/include /Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/components/mbedtls/mbedtls/library/x509_create.c
Traceback (most recent call last):
  File "/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/tools/ldgen/ldgen.py", line 158, in <module>
    main()
  File "/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/tools/ldgen/ldgen.py", line 134, in main
    mapping_rules = generation_model.generate_rules(sections_infos)
  File "/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/tools/ldgen/generation.py", line 352, in generate_rules
    self._create_exclusions(mapping_rules, default_rules, sections_infos)
  File "/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/tools/ldgen/generation.py", line 451, in _create_exclusions
    general_rule.add_exclusion(specific_rule, sections_info)
  File "/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/tools/ldgen/generation.py", line 100, in add_exclusion
    expansions = do_section_expansion(self, section)
  File "/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/tools/ldgen/generation.py", line 84, in do_section_expansion
    expansions = fnmatch.filter(sections_in_obj, section)
  File "/usr/local/Cellar/python@3.8/3.8.5/Frameworks/Python.framework/Versions/3.8/lib/python3.8/fnmatch.py", line 55, in filter
    for name in names:
TypeError: 'NoneType' object is not iterable
*** [.pio/build/esp32dev/esp32.project.ld] Error 1
```

valeros, 2020-12-14:
Hi @jonathandreyer ! Looks like you also need to add the object file extension used in the PlatformIO build system. Open `/Users/<USER>/.platformio/packages/framework-espidf@src-669852f02150559ed2825321cee4e6bd/tools/ldgen/generation.py` file at the [line 638](https://github.com/jonathandreyer/esp-idf/blob/release/v4.1/tools/ldgen/generation.py#L638) and modify it to something like this:
```python
...
if any(obj_key == obj + ext for ext in (".o", ".c.obj", ".c.o")):
...
```

jonathandreyer, 2020-12-14:
@valeros Thanks for your message. I have edited the file and that has fixed the error.

closed 2020-12-14 by jonathandreyer