platformio/platform-espressif32 #451: Toolchain xtensa32 missing for darwmin_arm64

clementnuss, 2020-12-14:
Hi,

The `toolchain_xtensa32` package is missing for the new Mac's with the M1 chipset (`darwin_arm64`).
I am trying to build it myself, then I will publish a binary in this issue.

It would help if I had access to build instructions for `toolchain_xtensa32`, does anyone have that?

Have a good week!

valeros, 2020-12-14:
Hi @clementnuss ! We appreciate that you want to help. The official instruction for compiling from sources can be found [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-setup-scratch.html#compile-the-toolchain-from-source).

valeros, 2022-04-18:
Toolchains package have been updated to support darwin_arm64.

closed 2022-04-18 by valeros