platformio/platform-espressif32 #452: Unable to build and upload SPIFFS filesystem image with esp-idf framework@2.1.0

jlu-battery-biz, 2020-12-17:
Hi everyone,

I am new to PlatformIO and I need some help on building and uploading SPIFFS filesystem image to my ESP32 board.

Let me throw my find first.

1. I am able to build and upload SPIFFS to my ESP32 board with Arduino framework
2. I am unable to build and upload SPIFFS to my ESP32 board with ESP-IDF framework

The Arduino project I created is following this link

[https://docs.platformio.org/en/latest/tutorials/espressif32/arduino_debugging_unit_testing.html](url)

Get started with Arduino and ESP32-DevKitC: debugging and unit testing —...
The ESP-IDF project I created is following this link

[https://docs.platformio.org/en/latest/tutorials/espressif32/espidf_debugging_unit_testing_analysis.html](url)

Get started with ESP-IDF and ESP32-DevKitC: debugging, unit testing, project...
For both project, I created a folder “data” and put the same file called “index.html”.
And here is my screenshot for ESP-IDF project

`Processing esp32doit-devkit-v1 (platform: espressif32@2.1.0; board: esp32doit-devkit-v1; framework: espidf)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (2.1.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40100.200827 (4.1.0)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
 - toolchain-xtensa32 2.80200.200827 (8.2.0)
Warning! Starting with ESP-IDF v4.0, new project structure is required: 
https://docs.platformio.org/en/latest/frameworks/espidf.html#project-structure
Reading CMake configuration...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 2 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <spiffs>
|   |-- <esp32_misc>
Building in release mode
================================================================================================== [FAILED] Took 2.28 seconds ==================================================================================================`

There is no error message telling me what exactly it is failing to build and upload to the ESP32.
I have no idea how to troubleshoot this issue.

Thank you in advance if someone could help me.
Best,
Leo

ul-gh, 2020-12-23:
Hi,

you can turn on verbose output using:

pio run -v > logfile.txt
or similar for a given environment: pio run -e release -v > logfile.txt
or set the setting: pio settings set force_verbose Yes

When the SPIFFS image is successfully built, you will find it from the project directory root under:
.pio/build/release/spiffs.bin. From that folder, when you have the esptool in your path, you can flash manually using:

```
esptool --chip esp32 --baud 921600 --before default_reset --after hard_reset \
    write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect \
    0x290000 spiffs.bin
´´´
==> The SPIFFS partition offset of 0x290000 is default IIRC for the Arduino environment. ESP-IDF uses a different partitioning scheme, you will have to use the correct offset.

valeros, 2021-01-21:
I fixed the build script, so it will print the error message properly. In a nutshell, the default ESP-IDF partition table doesn't have a SPIFFS section, so you will need to create your own partition table.

commit a57403c5 (2021-01-21): Print error message when there is no SPIFFS section in partition file
Resolve #452
closed 2021-01-21 by valeros