platformio/platform-espressif32 #454: ESP32 - Flashing a program using esptool (larger than 2MB)

lathareddy1976, 2020-12-18:
Hi,
We are trying to flash a program (larger than 2MB) in size. We have ESP32 module with 8MB flash and created the partition table as shown below:

When flashed the program using "pio run -t upload" command, the following error message is thrown.

We have modified the board file, but still we got the error. 

Where can we change the maximum flash size?

Zacharot, 2020-12-22:
app0,       app,  ota_0,      0x10000,    0x1E0000,
app1,       app,  ota_1,      0x1F0000,  0x1E0000,

First hex is the offset, second hex is the size. 1E0000 is 1,966,080

This partition would give you 2,949,120 (2D0000) for each. And 2,424,832 of spiffs (writable file system) for logs or whatever.

# Name,   Type, SubType, Offset,        Size,        Flags
nvs,          data, nvs,          0x9000,      0x5000,
otadata,   data, ota,          0xe000,      0x2000,
app0,       app,  ota_0,       0x10000,    0x2D0000,
app1,       app,  ota_1,       0x2E0000,  0x2D0000,
spiffs,      data, spiffs,        0x5B0000,  0x250000,

To break it down
Total:      8,388,608 bytes
boot:      36,864 - hence offset start at 0x9000
nvs:        20,480
otadata: 8,192
app0:     2,949,120
app1:     2,949,120
spiffs:     2,424,832

valeros, 2021-02-03:
Hi @lathareddy1976 ! Could you please share here your `platformio.ini`?

valeros, 2022-04-18:
Closing as no further info provided. Feel free to reopen the issue if you're still experiencing the problem with the latest upstream version of the platform.

closed 2022-04-18 by valeros