platformio/platform-espressif32 #455: Add support for ESP-IDF v4.2

septatrix, 2020-12-18:
This should resolve issues like #432 and similar ones. Also multiple bugs are fixed in v4.2 and new features added.

romocitto88, 2021-01-19:
@valeros  Can you give us any news about v4.2? Thank you in advance.

valeros, 2021-01-25:
Hi guys, I've pushed updates for ESP-IDF v4.2 please retest with the upstream version of the platform:
```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
framework = espidf
board = esp32dev
monitor_speed = 115200
```

alexreinert, 2021-01-26:
I just switched to that version and it seems, that there is an issue with board_build.embed_files.

I have multiple gziped files, which are included to be used in a webserver. Inside the webserver I use following code:

```
#define EMBED_HANDLER(_uri, _resource, _contentType)                   \
    extern const char _resource[] asm("_binary_" #_resource "_start"); \
    extern const size_t _resource##_length asm(#_resource "_length");  \
    esp_err_t _resource##_handler_func(httpd_req_t *req)               \
    {                                                                  \
        httpd_resp_set_type(req, _contentType);                        \
        httpd_resp_set_hdr(req, "Content-Encoding", "gzip");           \
        httpd_resp_send(req, _resource, _resource##_length);           \
        return ESP_OK;                                                 \
    };                                                                 \
    httpd_uri_t _resource##_handler = {                                \
        .uri = _uri,                                                   \
        .method = HTTP_GET,                                            \
        .handler = _resource##_handler_func,                           \
        .user_ctx = NULL};

EMBED_HANDLER("/*", index_html_gz, "text/html")
EMBED_HANDLER("/main.1e43358e.js", main_1e43358e_js_gz, "application/javascript")
EMBED_HANDLER("/main.1e43358e.css", main_1e43358e_css_gz, "text/css")
EMBED_HANDLER("/favicon.26242483.ico", favicon_26242483_ico_gz, "image/x-icon")
```

Worked before, but after the switch I get:

```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/hb-rf-eth.html
PLATFORM: Espressif 32 (2.1.0+sha.14e3f31) > HB-RF-ETH
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, esp32s2_kaluga_v1, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40200.210118 (4.2.0)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
 - toolchain-xtensa32 2.80400.210114 (8.4.0)
Reading CMake configuration...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Generating assembly for .pio\build\hb-rf-eth\main.1e43358e.css.gz.S
Generating assembly for .pio\build\hb-rf-eth\main.1e43358e.js.gz.S
Generating assembly for .pio\build\hb-rf-eth\index.html.gz.S
Generating assembly for .pio\build\hb-rf-eth\favicon.26242483.ico.gz.S
Linking .pio\build\hb-rf-eth\firmware_1_2_8.elf
c:/users/alex/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\hb-rf-eth\src\webui.o:(.literal._Z26index_html_gz_handler_funcP9httpd_req+0xc): undefined reference to `index_html_gz_length'
c:/users/alex/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\hb-rf-eth\src\webui.o:(.literal._Z26index_html_gz_handler_funcP9httpd_req+0x10): undefined reference to `_binary_index_html_gz_start'
c:/users/alex/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\hb-rf-eth\src\webui.o:(.literal._Z32main_1e43358e_js_gz_handler_funcP9httpd_req+0x4): undefined reference to `main_1e43358e_js_gz_length'
c:/users/alex/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\hb-rf-eth\src\webui.o:(.literal._Z32main_1e43358e_js_gz_handler_funcP9httpd_req+0x8): undefined reference to `_binary_main_1e43358e_js_gz_start'
c:/users/alex/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\hb-rf-eth\src\webui.o:(.literal._Z33main_1e43358e_css_gz_handler_funcP9httpd_req+0x4): undefined reference to `main_1e43358e_css_gz_length'
c:/users/alex/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\hb-rf-eth\src\webui.o:(.literal._Z33main_1e43358e_css_gz_handler_funcP9httpd_req+0x8): undefined reference to `_binary_main_1e43358e_css_gz_start'
c:/users/alex/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\hb-rf-eth\src\webui.o:(.literal._Z36favicon_26242483_ico_gz_handler_funcP9httpd_req+0x4): undefined reference to `favicon_26242483_ico_gz_length'
c:/users/alex/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\hb-rf-eth\src\webui.o:(.literal._Z36favicon_26242483_ico_gz_handler_funcP9httpd_req+0x8): undefined reference to `_binary_favicon_26242483_ico_gz_start'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\hb-rf-eth\firmware_1_2_8.elf] Error 1
```

Inside the build directory I see, that the *.S files are correctly generated, it just seems, that they are not linked in.

valeros, 2021-01-26:
Hi @alexreinert ! Have you declared that files in your CMakeLists.txt?

alexreinert, 2021-01-26:
> Have you declared that files in your CMakeLists.txt?

No, as this was not required according to https://docs.platformio.org/en/latest/platforms/espressif32.html#embedding-binary-data
As said, it worked before.

valeros, 2021-01-26:
The docs will be updated together with a new release of the espressif32 platform. At the moment, could you try to specify them as in this example https://github.com/platformio/platform-espressif32/blob/develop/examples/espidf-aws-iot/src/CMakeLists.txt#L8

alexreinert, 2021-01-26:
Could this be caused by the deletion of the line 
```
env.AppendUnique(PIOBUILDFILES=files)
```
in file `builder/frameworks/_embed_files.py`?

valeros, 2021-01-26:
That's right and it was intentional.

alexreinert, 2021-01-26:
It works, when I include it int eh CMakeLists.txt. I'm not that happy with this change, as I now need to maintain that files at two places.

misery, 2021-01-26:
Thanks for S2 support! I have some ESP-12k from AI-Thinker. Is that board compatible with that board configuration or is another cfg file necessary?

* https://docs.ai-thinker.com/en/esp32s2
* https://www.mischianti.org/2020/12/01/esp32-s2-pinout-specs-and-arduino-ide-configuration-1/

valeros, 2021-01-26:
@alexreinert Compatibility with the native ESP-IDF build system is our main priority. Starting with IDF v4.2 automatically generated files are now visible in the exported code model. Unfortunately, ESP-IDF has several different build steps where such files can be generated and has no mechanism to detect which custom command was used for generating that files. The best I can do is a to prepare a workaround that will skip generated files when compiling application sources. Sounds good?

@misery Seems like it should work via the `esp32-s2-saola-1` config, but feel free to submit a PR with a new manifest with correct information.

alexreinert, 2021-01-26:
> @alexreinert Compatibility with the native ESP-IDF build system is our main priority. Starting with IDF v4.2 automatically generated files are now visible in the exported code model. Unfortunately, ESP-IDF has several different build steps where such files can be generated and has no mechanism to detect which custom command was used for generating that files. The best I can do is to prepare a workaround that will skip generated files when compiling application sources. Sounds good?

Compatibility is a very good thing. I just do not understand, why the files need to be included using target_add_binary_data as well as board_build.embed_files. But I'm not deep inside the esp-idf toolchain, so if you say, that there are reasons for that, it would be fine for me. For now I added comments on both places to not forget, to change it on both places.
Maybe you should mention that fact in the release notes.

septatrix, 2021-02-01:
I will try to test the v3 release this week :D

valeros, 2021-02-05:
Support for ESP-IDF v4.2 is available in the latest release.

closed 2021-02-05 by valeros