platformio/platform-espressif32 #458: partitions.bin not built if PROGNAME is overridden

patrislav1, 2020-12-25:
I'm using a pre-build script that overrides the `PROGNAME` to rename the application binary file, as suggested in the [PlatformIO doc](https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#custom-firmware-program-name). Then PlatformIO will not build the `partitions.bin` anymore - even if I put a explicit `board_build.partitions` statement in my `platformio.ini` (see below). When trying to upload (`pio run -t upload`), esptool fails with `[Errno 2] No such file or directory: '(...)/.pio/build/nodemcu-32s/partitions.bin'`
```
progname = f"incinerator-{ver_txt}"
if "Import" in globals():
    # Set binary file name for PlatformIO build
    Import("env")
    env.Replace(PROGNAME=progname)
```    
When I comment out the `env.Replace(PROGNAME)` in my pre-build script, then the partitions.bin is built and esptool works again.

I am using PlatformIO version 5.0.3 and ESP32 Platform version 2.1.0.

```
[env:nodemcu-32s]
platform = espressif32
board = nodemcu-32s
framework = arduino
extra_scripts = pre:gitversion.py
upload_port = /dev/ttyUSB0
monitor_port = /dev/ttyUSB0
monitor_speed = 115200
board_build.partitions = default.csv
```

valeros, 2021-01-21:
Hi @patrislav1 ! It's strange, but I cannot reproduce the issue. The [dependency](https://github.com/espressif/arduino-esp32/blob/master/tools/platformio-build.py#L239) on the final firmware binary seems correct. Could you please compile and upload your project in verbose mode and and attach here the log?

valeros, 2021-02-19:
Please reopen if you still need help.

redakker, 2021-03-25:
Hi,
I have the same problem, I can reproduce all the time.
Steps:
- clean the build folder (build from zero)
- the modified PROGRAMNAME should contain "." (dot)
- use build and upload function (simple build is not enough)

Once you remove temporarily the dot from the modified filename and after a success build you put back the dot then it will solve the problem. It could be a workaround, but it is annoying. For example my code contains my version number, and I need to change my code because this issue

My PROGRAMNAME modification python code is:
`
revision = (
    subprocess.check_output(["git", "rev-list", "--count", "HEAD"])
    .strip()
    .decode("utf-8")
)
version = "undefined"

defFile = open("src/definitions.h")
for line in defFile:
    if line.startswith("#define MAJOR_VERSION"):
        version = line.replace("#define MAJOR_VERSION ", "")
        version = version.strip()
defFile.close()

program_file_name = "24H_v" + version + "-" + revision 

env.Replace(PROGNAME=program_file_name)
`

MAJOR_VERSION contains the dot. For example: 0.9

Final generated filaname is: Uploading .pio\build\esp32dev\24H_v0.99-36.bin

And the result of the build fail:
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'D:\\........\\projects\\esp24h\\.pio\\build\\esp32dev\\partitions.bin'
*** [upload] Error 2

I've checked the build folder. There should be a .elf file too. But if the PROGRAMNAME contains a dot ".elf" extension is missing
I got 24H_v0.99-36 instead of 24H_v0.99-36.elf

makerpat, 2021-04-07:
Same problem here. No **partitions.bin** in env-folder. And an ".elf" file without extension.

valeros, 2021-04-07:
Resolved in https://github.com/platformio/platformio-core/issues/3906. Please upgrade your PlatformIO to the latest from the dev branch via the `pio upgrade --dev` command.

closed 2021-04-07 by valeros