platformio/platform-espressif32 #470: Could not initialize project

Tinkot, 2021-01-20:
PIO Core Call Error: "The current working directory /Users/tjeerdvandijk/Documents/HAN/IOT_Project/svn/branches/development/Onderzoek Timers/test will be used for the project.\n\nThe next files/directories have been created in /Users/tjeerdvandijk/Documents/HAN/IOT_Project/svn/branches/development/Onderzoek Timers/test\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Processing esp32dev (platform: espressif32; board: esp32dev; framework: espidf)\n--------------------------------------------------------------------------------\nVerbose mode can be enabled via `-v, --verbose` option\nCONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html\nPLATFORM: Espressif 32 (1.12.4) > Espressif ESP32 Dev Module\nHARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash\nDEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)\nPACKAGES: \n - framework-espidf 3.40001.200521 (4.0.1) \n - tool-cmake 3.16.4 \n - tool-esptoolpy 1.20600.0 (2.6.0) \n - tool-ninja 1.9.0 \n - toolchain-esp32ulp 1.22851.191205 (2.28.51) \n - toolchain-xtensa32 2.80200.200827 (8.2.0)\nError: Detected a whitespace character in project paths.\n========================== [FAILED] Took 0.76 seconds =========================="

valeros, 2021-01-20:
You have white space characters in project path `Onderzoek Timers`, starting with `ESP-IDF v4.0` they  are not allowed.  More info [here](https://docs.platformio.org/en/latest/frameworks/espidf.html#configuration).

closed 2021-01-20 by valeros

TNeutron, 2023-09-13:
Very annoying bug. Still pain in the ass even after 3 years of issuing.

valeros, 2023-09-13:
@TNeutron this limitation is imposed by the underlying IDF build system:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html#start-a-project
```
The ESP-IDF build system does not support spaces in the paths to either ESP-IDF or to projects.
```

humid-sirocco, 2024-01-22:
Looks like they may have fixed this: https://github.com/espressif/esp-idf/issues/5576

Gisbert12843, 2024-01-27:
Bug is not resolved. Can not create any projects. Even with the Project Name "Test", Location set as C:/test, Folder opened via VS Code being "C:/Test, Board being ESP32 C3 Dev KitC 02 (doesnt matter tried various) + Framework ESPIDF

Error:

PIO Core Call Error: "The following files/directories have been created in C:\\test\\test\\test\r\ninclude - Put project header files here\r\nlib - Put project specific (private) libraries here\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\nResolving esp32-c3-devkitc-02 dependencies...\r\nAlready up-to-date.\r\nUpdating metadata for the vscode IDE...\r\n\n\nUserSideException: Processing esp32-c3-devkitc-02 (platform: espressif32; board: esp32-c3-devkitc-02; framework: espidf)\r\n--------------------------------------------------------------------------------\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitc-02.html\r\nPLATFORM: Espressif 32 (6.5.0) > Espressif ESP32-C3-DevKitC-02\r\nHARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash\r\nDEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)\r\nPACKAGES: \r\n - framework-espidf @ 3.50102.240122 (5.1.2) \r\n - tool-cmake @ 3.16.4 \r\n - tool-esptoolpy @ 1.40501.0 (4.5.1) \r\n - tool-idf @ 1.0.1 \r\n - tool-mconf @ 1.4060000.20190628 (406.0.0) \r\n - tool-ninja @ 1.9.0 \r\n - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823 \r\n - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208 \r\n - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) \r\n - toolchain-riscv32-esp @ 12.2.0+20230208\r\nError: Detected a whitespace character in project paths.\r\n========================== [FAILED] Took 0.87 seconds =========================="

hansratzinger, 2024-03-18:
Temporary solution listed on top is **NOT WORKING !!!!!!!!!**  I removed .platformio directory in my user directory and started new. Is has been again located in my user account and not in C:\

wangyinggang-97, 2024-07-06:
thank very much

palmedic, 2024-07-24:
I just made sure that my project's NAME has no white space

weinerschizel, 2024-08-13:
Experiencing same issue. Cannot remove white space from project name as it's an example project.

milotruck-shin, 2024-10-26:
Hi, is there a fix for this yet? I dont have whitespace in my project name and Ive also followed the solution

milotruck-shin, 2024-10-26:
> Still an issue as of ESP-IDF v5.3.1. I'm getting mixed information that they now support project directories with spaces, but everyone else keeps saying they don't. Moving the `.platformio` core directory to the `C:\` drive also does not fix this.

refer to my comment above, i hope it helps